timmbogner/Farm-Data-Relay-System #1: Maximum distance in the field

garudaonekh, 2021-06-30:
hi,
How far can espnow reach in case of internal antenna and as well as external antenna? I am currently using Tasmota with trouble with range in the open field.

I mainly use esp32 wroom 32U and esp8266 d1

garudaonekh, 2021-06-30:
By the way, have you tried LR mode(long range mode) of esp32? I tried it but the range is dissapointing, i can Only reach 60m

timmbogner, 2021-07-01:
I haven't done many range tests yet, but an early version of the system used to span about 120 meters of open field pretty easily.
I haven't tested LR mode very thoroughly. Since it isn't compatible with 8266 I've been hesitant to bother.

garudaonekh, 2021-07-01:
120m is impressive. what kind of antenna used?and which board?

timmbogner, 2021-07-01:
Believe it or not, it was just an ESP-01. I currently use WeMos modules generally and an ESP32 DevKit for the terminal module.

garudaonekh, 2021-07-01:
Let me try it using ESP32-WROOM 32U(External antenna) and see how far it can reach.
I plan to use this module for the sensor(https://github.com/LilyGO/LILYGO-T-OI)
and ESP32-WROOM 32U as gateway

timmbogner, 2021-07-02:
https://youtu.be/oz0a7Ur7nko
Here's a good video with range tests.

closed 2021-07-06 by timmbogner