timmbogner/Farm-Data-Relay-System #6: Two way communication of the gateway

garudaonekh, 2022-04-22:
Hi, can the gateway pass data back from the controller to devices. For example, sensor config data and relays.

timmbogner, 2022-04-28:
Yes, there are a few ways to achieve this using FDRS. While these methods are undocumented and subject to change, they work in some situations.
The overall technique (for now) is to use the DataReading.id as the device's address and have the device check every packet it receives for an id that matches its own. In this way, you just need to forward all traffic to this device and let it sort out the rest. 
To send a packet from the MQTT gateway to a motor sketch for instance, start by adding an MQTT action to the first gateway:
`#define MQTT_ACT sendSerial();`. The gateway will now subscribe to 'esp/fdrs' and send any data it receives over the serial port.
Next, you will set the first ESP-NOW gateway to transmit all serial data to either broadcast ESP-NOW: `#define SERIAL_ACT bufferESPNOW(0);` This is a somewhat hidden method of sending a broadcast packet using FDRS. If there are other gateways listening for general ESP-NOW packets, they will treat these packets as sensor readings, thus this is an imperfect method for larger setups.
The other way is to set the MAC address of the device you are controlling to have the same prefix as your gateways and a unique final digit (0x42 for example), then direct traffic to it with:`#define SERIAL_ACT sendESPNOW(0x42);`.
 
I'm hoping to make this more elegant in the future, but the system is definitely designed with 2-way communication in mind.

garudaonekh, 2022-05-03:
Based on what you said, will it be able to scale to 200 sensors or above?

closed 2022-06-17 by timmbogner