timmbogner/Farm-Data-Relay-System #25: UART To PI MQTT Local Loopback

Devilbinder, 2022-06-27:
Looking at the setup diagram it is possible for a ESP to get connected directly to a PI and pull the serial JSON . The serial JSON can then be translated to MQTT and sent to the broker on the local loop back on the PI itself. This will eliminate the need for the extra ESP at the PI.
[image]

timmbogner, 2022-06-27:
This was an intermediate solution that I used for a good portion of time! It used to be the go-to method, but Andreas pointed out that it would be really beneficial to put the ESP-NOW antenna in an elevated and possibly obscure position, which wouldn't be a good spot for the server(s). 
I still suggest this to people with more critical applications.
At one point it was officially documented, but it seems to have been removed in my last rewrite of the gateway readme.  Noted.

timmbogner, 2022-06-27:
[image]

timmbogner, 2022-06-27:
[image]

lekrom, 2022-06-28:
One can still implement this scheme (getting rid of the final ESP) and have the antenna at an elevated (or otherwise remote) location if needed. In such a case, I would recommend using RS-232 or even RS-485 (for more distance) for the link between the ESP and the PI instead of TTL-serial. Cheap TTL to RS-232 modules are available everywhere and easy to build as well.

Devilbinder, 2022-06-28:
Would recommend RS485. Had enough industrial equipment on farms cause issues from EMI. Bi-directional communication becomes a pain then. RS422 is bi-directional but the it is sometimes expensive basically 2x RS485.

closed 2022-10-02 by timmbogner