timmbogner/Farm-Data-Relay-System #30: SPI.begin() does not take arguments

thefeiter, 2022-06-30:
When using the LoRa Sensor the compiler throws an Error, because the SPI.begin() does not take the used pins as arguments.

It is required to use the standard SPI pins of the board I am using.

Is there I differend SPI lib I have to use?

Devilbinder, 2022-06-30:
What board are you compiling with?

closed 2022-06-30 by thefeiter

timmbogner, 2022-07-01:
> even as I dont't have an application atm. I'm just twiddling around.

Thanks, I can relate! I don't even work for the farm that I designed this for anymore, so now I'm just having fun and hoping others do the same. At first I envisioned the system like a "series of tubes"😁 that pushed data from end-to-end like a fluid. As I added more flexibility I began thinking of it as a model railroad set, where the user can set up a "track" for the data to run on. It's really more fun than one might think!