timmbogner/Farm-Data-Relay-System #39: General Gateway: LORA_ACT vs. LORA_GET and LORA_SEND

Gulpman, 2022-07-02:
I have a question regarding the new general gateway in dev-branch Examples 

In the v1 version there was a definition of what has to be done with a package arriving on a specific protocol – that was pretty straight forward, e.G.:

`#define LORAG_ACT sendSerial();`

So incoming lora trafic goes to serial. Check!

In the new version we have two defines:
```
//Where we get the data from
#define LORA_GET
#define MQTT_GET
#define ESP_GET
#define SER_GET 

//Where we send the data to
#define LORA_SEND
#define MQTT_SEND
#define ESP_SEND
#define SER_SEND 
```

So there is a <PROTOCOL>_GET and <PROTOCOL>_SEND. 
I think everything not commented out is generally supported and initialized. Check!
However, how do I do the mapping from incoming protocol to outgoing protocol? For example as in the old version, how can I manage it that LORA incoming traffic should be routed to serial out?

I don’t get it. ☹ Any help is appreciated.

Devilbinder, 2022-07-03:
Was not really intended as the final config solution.

So the _GET tag determines witch protocols collects incoming data. The _SEND tag determines witch protocols can send data out on the network.

So for example if you have Lora as your incoming data and Serial as your outgoing data you would define.
`#define LORA_GET` and `#define SER_SEND`

This get the data from the lora module and stores it.
https://github.com/timmbogner/Farm-Data-Relay-System/blob/7bf774ddc9a21874bc00b8494e62fff963e405aa/Examples/Universal_Gateway/Universal_Gateway.ino#L128

This makes a copy of the lora data that a specific peer has sent.
https://github.com/timmbogner/Farm-Data-Relay-System/blob/7bf774ddc9a21874bc00b8494e62fff963e405aa/Examples/Universal_Gateway/Universal_Gateway.ino#L160

The copy data can now be sent to any of the send endpoints in this case the Serial. 
https://github.com/timmbogner/Farm-Data-Relay-System/blob/7bf774ddc9a21874bc00b8494e62fff963e405aa/Examples/Universal_Gateway/Universal_Gateway.ino#L192

In this example `espnow_peer_2_data` would get replaced with  `lora_peer_1_data`.

The internal store data for the Lora then gets deleted for that particular peer.
https://github.com/timmbogner/Farm-Data-Relay-System/blob/7bf774ddc9a21874bc00b8494e62fff963e405aa/Examples/Universal_Gateway/Universal_Gateway.ino#L225

TL;DR
This lets you pipe the data to whatever you want.

timmbogner, 2022-07-04:
@Devilbinder Sorry! I hadn't realized you had the piping method sorted out. I'll try it soon and give you some feedback, but I'm going to keep developing the current gateway as well.

Devilbinder, 2022-07-08:
Wrote a small use example here without any configs.

https://github.com/timmbogner/Farm-Data-Relay-System/pull/29#issuecomment-1179386143

Devilbinder, 2022-07-11:
@Gulpman As per your request on #29. LoRa to LoRa is still listed as a `TODO`.
https://github.com/timmbogner/Farm-Data-Relay-System/blob/7bf774ddc9a21874bc00b8494e62fff963e405aa/Examples/Universal_Gateway/fdrs_gateway.cpp#L657
```cpp
#include "fdrs_gateway.h"
#include "fdrs_config.h"

uint8_t selfAddress[6] =   {MAC_PREFIX, UNIT_MAC};

uint8_t LoRa1[6] =         {MAC_PREFIX, LORA_PEER_1};
uint8_t LoRa2[6] =         {MAC_PREFIX, LORA_PEER_2};
std::vector<DataReading_t> lora_peer_1_data;
std::vector<DataReading_t> lora_peer_2_data;

#ifdef ESP32
Serial_FDRSGateWay SerialGW(&Serial1,115200);
#else
Serial_FDRSGateWay SerialGW(&Serial,115200);
#endif

std::vector<DataReading_t> serial_data;
LoRa_FDRSGateWay LoRaGW(MISO,MOSI,SCK,SS,RST,DIO0,BAND,SF);

void setup() {

#ifdef ESP32
  SerialGW.init(SERIAL_8N1,RXD2,TXD2);
#else
  SerialGW.init();
#endif
  LoRaGW.init(selfAddress);
}

void loop() {

//collect data
  LoRaGW.get();

//get the collected data
  lora_peer_1_data = LoRaGW.get_peer_data(LoRa1);
  lora_peer_2_data = LoRaGW.get_peer_data(LoRa2);
  
//send the data
  SerialGW.release(lora_peer_1_data);
  SerialGW.release(lora_peer_2_data);

//clear out the buffers
  SerialGW.flush();
  LoRaGW.flush(LoRa1);
  LoRaGW.flush(LoRa2);

  delay(1000);

}
```

closed 2022-08-01 by timmbogner