timmbogner/Farm-Data-Relay-System #47: DEBUG define with PlatformIO will cause build to fail

aviateur17, 2022-07-07:
Probably not to concerned about PlatformIO support but if compiling the project with PlatformIO and using the #define DEBUG in fdrs_functions.h and fdrs_config.h the build will fail:

In file included from /home/xxxx/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/newlib/stdio.h:47:0,
                 from /home/xxxx/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/esp32/esp_err.h:17,
                 from /home/xxxx/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/esp32/esp_now.h:19,
                 from src/main.cpp:9:
/home/xxxxx/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/newlib/sys/reent.h:463:10: error: #if with no expression

Changing DEBUG to something else in both of those files seems to resolve the issue.

thefeiter, 2022-07-10:
:warning: This has to be mentioned in a changelog. It caused some confusion for me, since no debug messages were sent because I copied the fdrs_config file.

aviateur17, 2022-07-12:
Resolved in dev branch to be merged with main.

closed 2022-07-12 by aviateur17