timmbogner/Farm-Data-Relay-System #59: Compile error: Help needed with Universal_Sensor_beta - choose protocol

Gulpman, 2022-07-12:
I wanted to clean-up the configuration for the Universal_Sensor_beta. My intention was, that the user should not edit the class files but only the sensor sketch itself (.ino file) or the specific configuration file (fdrs_sensor_config.h).

In the current implementation of Universal_Sensor_beta the protocol being used can be activated by uncommenting one or the other

```c++
#define USE_LORA  
//#define USE_ESPNOW
```
within `Farm-Data-Relay-System\examples\Universal_Sensor_beta\fdrs_sensor.h`

Therefore I wanted to move the declaration, which protocoll the Universal_Sensor is supposed to use to the sensor configuration file:

Actually it is defined within fdrs_sensor.h, which compiles without issues.
```c++
[...]
#if ENABLE_DEBUG == 1
#ifndef FDRS_DEBUG
#define FDRS_DEBUG
#endif
#endif

#define USE_LORA  

#if defined(ESP8266)
#include <ESP8266WiFi.h>
#include <espnow.h>
#elif defined(ESP32)
[...]
```

The end user shouldn't edit fdrs_sensor.h. So I wanted to move the #define to fdrs_sensor_config.h:

```c++
[...]
#define READING_ID    1   	//Unique ID for this sensor
#define GTWY_MAC      0x04 	//Address of the nearest gateway

#define DEEP_SLEEP
//#define POWER_CTRL    14

// Uncomment the sensor type you want to use 
#define USE_LORA
//#define USE_ESPNOW

//Pins for UART data interface (ESP32 only)
#define RXD2 14
#define TXD2 15
[...]
```

But this leads to a compilation error:
```
sketch\Universal_Sensor_beta.ino.cpp.o:(.literal.startup._GLOBAL__sub_I_FDRS+0x8): undefined reference to `FDRSLoRa::FDRSLoRa(unsigned char, unsigned short, unsigned char, unsigned char, unsigned char, unsigned char, unsigned char, unsigned char, unsigned int, unsigned char)'
sketch\Universal_Sensor_beta.ino.cpp.o:(.literal.exit._GLOBAL__sub_D_FDRS+0x0): undefined reference to `vtable for FDRSLoRa'
sketch\Universal_Sensor_beta.ino.cpp.o: In function `_GLOBAL__sub_I_FDRS':
D:\dev\Arduino\libraries\Farm-Data-Relay-System\examples\Universal_Sensor_beta/Universal_Sensor_beta.ino:16: undefined reference to `FDRSLoRa::FDRSLoRa(unsigned char, unsigned short, unsigned char, unsigned char, unsigned char, unsigned char, unsigned char, unsigned char, unsigned int, unsigned char)'
collect2.exe: error: ld returned 1 exit status

```

I tried to do the define as first statement in the ino-file, to be 100% sure it is defined prior to any use of it but it throws the same compiler error.

I also tried to put the sketch and all includes into the root directory and into the src directory (in a temporary library, to be sure ther is no interference with anything else but I do still get the compile error.

@Devilbinder or anyone else: Can someone explain why the #define must be in the fdrs_sensor.h? And what can I do, to move it to the sketch or fdrs_sensor_config.h? Both of them should be known to fdrs_sensor.h or am I getting something wrong here?

Gulpman, 2022-07-12:
I tried to find out what's going on and searched for "undefined reference to vtable for" and stumbled upon this article: https://stackoverflow.com/questions/3065154/undefined-reference-to-vtable

User iammilind says: 

> another Qt related problem is that if the file with Q_OBJECT is copied externally, but not yet part of the .pro file, then though it compiles fine, it doesn't link. We have to add that .h/.cpp file into the .pro file to be able to qmake. 

Though this is Arduino and not Qt I thought it would be worth a try - as the issue is comming from the object file (FDRS_Universal_Sensor.ino.o)

I added 
```c++
#include "fdrs_sensor.cpp" 
```
below the corresponding header file into the Sensor sketch and - voila, it compiles! 

But... now I'm getting a new compiler error:

```
In file included from F:\dev\FDRS_UniversalSensor\FDRS_UniversalSensor.ino:18:0:
D:\dev\Arduino\libraries\Farm-Data-Relay-System\src/FDRS_Sensor/fdrs_sensor.cpp: In member function 'void FDRSBase::sleep(int)':
D:\dev\Arduino\libraries\Farm-Data-Relay-System\src/FDRS_Sensor/fdrs_sensor.cpp:69:33: error: 'sleep_time' was not declared in this scope
   esp_sleep_enable_timer_wakeup(sleep_time * 1000000);
                                 ^
```
Which makes sense for me, as I couldn't find a place, where `sleep_time` is actually defined. But then I am wondering why it compiles flawlessly, if the `#define USE_LORA` is done in `fdrs_sensor.h` file...

So still I do not understand what's going on.

Anyone who can help?

timmbogner, 2022-07-13:
You tried installing and switching the "includes" to <includes>? That is my only thought. I assume Binder will have some insight...

Gulpman, 2022-07-13:
> You tried installing and switching the "includes" to <includes>? That is my only thought. I assume Binder will have some insight...

I think so - actually I tried out all combinations I think. I just stepped one step back again and try to find a viable info about the include order of the Arduino IDE. Will share if I found out more.

aviateur17, 2022-07-14:
@Gulpman, I get the same error.  Here's what I did to resolve.  I'm no expert but the issue must be the order in which the compile interprets the files.

Here's what I did and compiles properly on my Linux system using Arduino 1.8.19

1. In Universal_Sensor_Beta.ino -- Comment out line 15 - #include "fdrs_sensor_config.h"
2. In fdrs_sensor.h -- Add a line 14 - #include "fdrs_sensor_config.h"
3. In fdrs_sensor.h - Line 25 comment out #define USE_LORA
4. In fdrs_sensor_config.h - Line 20 - uncomment #define USE_LORA

Those steps seem to result in a successful compilation.

Gulpman, 2022-07-14:
> @Gulpman, I get the same error. Here's what I did to resolve. I'm no expert but the issue must be the order in which the compile interprets the files.

Definitively! I have read so many tutorials/documentations telling totally different stuff about how ArduinoIDE is handling compiling and linking... The consens was to use something different as the way Arduino IDE is handling this is somehow unpredictable. 

OT: I would like to try out Platform.IO. From what I read, these kind of issues are explaind better when arising so fixing should be easier. As I think you are using Platform.IO (see #47  ;) ) - can you give a short summary on how to import this repo there? I tried to import it via the import dialogue but it complains that there is no .ino file.

> Here's what I did and compiles properly on my Linux system using Arduino 1.8.19
> [...]
> Those steps seem to result in a successful compilation.

Works! Thank you very much @aviateur17 

I was playing around with the folder structure of the library in my local copy as I stumbled upon the Arduino library specification on my hunt for the root case as well. Reason is, that I'm unhappy with the fact, that each example holds the library files. I think they should be included from the library path (root dir or from a src subdir). I will need to test that approach a little bit more but overall I think this is something which must be done (get rid of redundancy) to have a clean project. Unfortunately I'm a noob on these things so I will not change it until I got it in my brain what the correct way to do it is. :)

But this topic can be closed - will open a new one on the structure if I know better. 

Thanks again @aviateur17

Gulpman, 2022-07-14:
Resolved and therefore closed. 👍

aviateur17, 2022-07-14:
Glad it worked for you!   Going to VSCode and PlatformIO is a lot of work but well worth it in my opinion.  I'm going on a week long holiday soon and so I don't have time to type up everything right now but I can do that in a week or so.

Basically
1. Download and Install VSCode
2. Install a bunch of extensions including PlatformIO
3. In PlatformIO install ESP32 board and download MQTT, LoRa, and other libraries
4. Take the files from this repo and copy them into the project directories
5. Create a platformio.ini file
6. Open the project and work and compile

commit 3878c277 (2022-07-15): Fixed #59

Fixed #59 with the help of aviateur17. Explanation here: https://github.com/timmbogner/Farm-Data-Relay-System/issues/59#issuecomment-1183915731
closed 2022-07-15 by timmbogner