timmbogner/Farm-Data-Relay-System #60: LoRa doesn't seem to work in gateway in main branch

aviateur17, 2022-07-12:
Couldn't get LoRa to work in the gateway code in the main branch until I added the lines below to setup() in the gateway code...

LoRa.setPins(FDRS_SS, FDRS_RST, FDRS_DIO0);
LoRa.begin(FDRS_BAND);

Note I am using PlatformIO so I had to rename the defines in order for the code to compile.

timmbogner, 2022-07-12:
Ope, yep it is time to pull the 'dev' branch to main... Does the gateway in 'dev' branch work okay in PlatformIO? It's about time I went down that path myself...

aviateur17, 2022-07-12:
> Ope, yep it is time to pull the 'dev' branch to main... Does the gateway in 'dev' branch work okay in PlatformIO? It's about time I went down that path myself...

Yes, gateway code in dev branch works as far as receiving LoRa is concerned.   One question though, and I can make this a different issue if need be, but why are we using an external NTP library when ESP supports NTP via native libraries?  Seems this project is geared towards ESP8266 and ESP32 anyways?

timmbogner, 2022-07-12:
Is it native on 8266 as well?
@thefeiter Might want to look into this.

aviateur17, 2022-07-12:
> Is it native on 8266 as well? @thefeiter Might want to look into this.

Seems to use the sntp library and time.h

https://github.com/espressif/ESP8266_RTOS_SDK/blob/a192988b7906680440213eefc730d99dec0ce237/examples/protocols/sntp/main/sntp_example_main.c

aviateur17, 2022-07-12:
LoRa gateway receive resolved in dev branch to be merged in main.

closed 2022-07-12 by aviateur17

thefeiter, 2022-07-12:
> Is it native on 8266 as well? @thefeiter Might want to look into this.

Seems like it is.

> why are we using an external NTP library when ESP supports NTP via native libraries

Main reason is: I didn't know about the native libraries, since they didn't show up on my search for NTP time sync for esps.