timmbogner/Farm-Data-Relay-System #61: OTA updates

aviateur17, 2022-07-12:
Would be good to support Over The Air updates for the MQTT gateway (as that has WiFi enabled by default) and eventually the UART gateway and remote devices.   For the UART gateway and remote devices maybe have a button ( would need to be added and could support other functionality in addition) that, if held down for 5 seconds or so, would enable WiFi and receive OTA updates?

Thoughts?

thefeiter, 2022-07-12:
OTA updates are quite powerful, I use them for the ESPs in my home.

But I think the Sensors, once working should be fine without OTA updates. And if you go to them to press a Button for updates, you could also be standing there with your laptop for 30 seconds to update via serial.

Also the WIFi range is not that big so you would have to bring the node closer to your router to update (assuming you have a more distributed network in the field).

timmbogner, 2022-07-12:
I mean, actually with sensors I guess you are free to add whatever capability you like since it's just a library. Especially since it sounds like it would be a fully separate mode. I just don't think it's within the scope of FDRS to handle. Maybe for MQTT mode if it's super-unobtrusive.

aviateur17, 2022-07-13:
Low priority issue to be taken up later after higher priority jobs are completed.

closed 2022-07-13 by aviateur17