timmbogner/Farm-Data-Relay-System #75: enhancement:  buffer data in sensor if gateway not reachable?

aviateur17, 2022-07-25:
If sensor tries to send data to gateway and no ack is received then buffer data in memory until either memory is full or data is acked instead of just discarding data?

timmbogner, 2022-07-25:
A buffer would only be helpful with timestamps, and then you'd have to add timestamps to the DataReading type to send over the air. This would require too much extra stuff for right now.
Keeping time more precisely is a struggle without an RTC. I just talked to someone trying to take readings once a day, and the time-drift is shifting their reading by ~40 seconds.
Once two-way communication is established there will be a way to pump current timestamps out to both sensors and controllers. That will open up lots of possibilities!

aviateur17, 2022-07-25:
This issue has a dependency upon two way communication and sending timestamps to sensor node and including them in the DataReading structs.  Will close this now and revisit later after two way communication is enhanced with SystemPackets.

timmbogner, 2022-09-09:
@aviateur17 
Hey, I'm really sorry you haven't been around. I definitely could have handled things better. I hope with some of my latest changes you can see what I was so worried about with physical addressing.
I really appreciate everything you've done to help. If you decide to to jump back in, you'll be most welcome! I plan to make the next release with two-way support limited to ESP-NOW only, and work on LoRa over the winter (LoRa gateways can still/already pass two ways though). I could definitely use your help with that.
Feel free to shoot me an email.

aviateur17, 2022-09-13:
Hey Timm, I have been away for awhile as other stuff has come up in life.  I've been lurking and expect that I will have some time to help out if I can with the project.  I think you've handled the project just fine.  Everyone has their own way that they do things and it's great that you have a vision for the project.   I'm excited to see where it goes and help when I can.