timmbogner/Farm-Data-Relay-System #78: Discussion - compilation with Arduino - #includes

aviateur17, 2022-07-26:
I assume you guys are using Arduino to compile and test?  Are you getting these missing library errors where you need to change your #includes to have parenthesis (") instead of greater than or less than (<   >)?

```
#include "fdrs_globals.h"
```
vs
```
#include <fdrs_globals.h>
```

When I try to compile with Arduino there are several I need to change to parenthesis in order for compilation to work.  Wonder what you guys are doing differently?

timmbogner, 2022-07-26:
Quotes indicate the file is in the local sketch folder, carats (<>) are for a file in the Arduino libraries folder.

aviateur17, 2022-07-26:
> Quotes indicate the file is in the local sketch folder, carats (<>) are for a file in the Arduino libraries folder.

Thanks, yes I knew that.  I guess my question was are you copying the header files into the sketch folder or are you putting them into the libraries folder?  When I compile myself I just copy them all to the sketch folder.

aviateur17, 2022-07-26:
I guess since they are all encapsulated by < and > that they are in the libraries folder.  Thanks!

closed 2022-07-26 by aviateur17