timmbogner/Farm-Data-Relay-System #85: Image transfer from ESP32-Cam

garudaonekh, 2022-07-30:
Hi, is it feasible to send image thru EspNow?
My scenario is to send time lapse photo of vegetable three time a day to observe vegetable progress

timmbogner, 2022-07-30:
The simple answer: this is currently beyond the scope of the project, as images require much more data to be sent through the system.

However, I hope to continue designing FDRS in a way that will facilitate this in the future. I'm going to use this opportunity to talk about how data moves and how we'll describe it.

People tend to jump to the conclusion that FDRS is a mesh topology when they first read about it. It fills the same void that meshes do, getting data from the "edge". In the video, Andreas described it as "star", which is much more accurate. I'm still not sure it fully describes the concept I have behind FDRS, though. In FDRS, data is entered in one place, then moves down a preset path until it has nowhere left to go. It can be looked at/copied along the way, but on the lowest level there is no concept of an end-destination for any data. To describe this, I propose using the term "wormhole topology" as it resembles such structures from Star Trek.

But yeah, in the future: a new packet type for bulk data with frames and such so that a file can be sent over the course of multiple packets. This is pretty far down the line, though.

garudaonekh, 2022-07-30:
Ok, I understand now. I see a github project which transfer image via EspNow to another EsP32 but it lacks what your project provides.

garudaonekh, 2022-08-12:
I have used LORA to transfer small size image(around  50kB). I breakdown into 255 packet size. And the sender will resend broken packet. It took around 3 minutes. So I think if I breakdown like this on FDRS, is it feasible or need a lot of effort?

timmbogner, 2022-08-12:
Yes, this will be the basic idea. Currently FDRS sends data in little chunks which are great for individual readings, but would be wasteful to use for a large amount of data. Essentially, files would use the same transportation methods, but files would be split into ~245 byte chunks. The chunks would be sent with an ID and frame number attached, so as to be reassembled at the front-end.

timmbogner, 2023-03-04:
Idea pinned for future development in https://github.com/timmbogner/Farm-Data-Relay-System/discussions/148.

closed 2023-03-04 by timmbogner