timmbogner/Farm-Data-Relay-System #90: Compiling error with LoRa in Arduino

timmbogner, 2022-08-03:
Just got this while reorganizing. I can't seem to tag you properly, Jeff, but I assume you'll see this 🙃
>sketch\4_UART_Gateway.ino.cpp.o:(.literal._Z7getLoRav+0x90): undefined reference to `transmitLoRa(unsigned short*, SystemPacket*, unsigned char)'
sketch\4_UART_Gateway.ino.cpp.o: In function `getLoRa()':
C:\Users\webma\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Misc/SafeBoolIdiom.hpp:15: undefined reference to `transmitLoRa(unsigned short*, SystemPacket*, unsigned char)'
C:\Users\webma\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Misc/SafeBoolIdiom.hpp:15: undefined reference to `transmitLoRa(unsigned short*, SystemPacket*, unsigned char)'
C:\Users\webma\OneDrive\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Misc/SafeBoolIdiom.hpp:15: undefined reference to `transmitLoRa(unsigned short*, SystemPacket*, unsigned char)'
collect2.exe: error: ld returned 1 exit status

BTW, if I'm not mistaken you followed that note I made in the code about making the transmitLoRa() function equivalent to esp_now_send(). Much appreciated!

aviateur17, 2022-08-03:
Should have been resolved in PR #88 commit fdc941658ec3ab7d08dee555ef6d00dc8367fe80.   Try adding function prototypes near the top of the file.
```
// Function prototypes
void transmitLoRa(uint16_t*, DataReading*, uint8_t);
void transmitLoRa(uint16_t*, SystemPacket*, uint8_t);
static uint16_t crc16_update(uint16_t, uint8_t);
```

aviateur17, 2022-08-03:
As in the other issue, it seems somehow the void transmitLoRa(uint16_t*, SystemPacket*, uint8_t); function itself was missing. Hopefully adding it back to fdrs_functions.h resolves the issue.

timmbogner, 2022-08-03:
I believe it is fixed now. If I were a betting man, I'd bet this was my own fault. Thanks for your patience!

closed 2022-08-03 by timmbogner