timmbogner/Farm-Data-Relay-System #93: ESP-01 Relay Controller question

garudaonekh, 2022-08-10:
Hi,
Currently I am controlling my GPRS board thru SMS. What I plan to do is to:
1-Connect ESP-01 to my GPRS board thru UART
2- The ESP-01 act as gateway to deliver my SMS thru UART and send out ESPNOW message to the nodes.
3-  The nodes here are ESP-01 connected with relay board(1 channel or 2 channel relay) and the node will open/close relay based on ESP-Now message from  ESP01 that's connected to my GPRS board.

How should I implement your FDRS? Maybe your Irrigation example best suit my scenario. But I see your code can't transfer thru UART on esp8266. Please note that I use ESP-01 because it's UART and easy to plug into my controller and easy for end users to install it.

timmbogner, 2022-08-11:
Hello!
I am getting very close to completing a new "official" method for sending controller packets. I should have it completed within a week. If you don't mind waiting a few days, you can be the first to test it! 

Unless I've made a mistake, there should be no limitation when using UART on 8266. You may want to disable FDRS_DEBUG to keep it from sending serial debug messages to the MQTT gateway, however.

garudaonekh, 2022-08-12:
For switch, we need acknowledgement packet. do you have this in your official release?

timmbogner, 2022-08-12:
Yes, it will be there in some form. A method for LoRa packet confirmation has been completed in the 'dev' branch. On ESP-NOW I will probably utilize the internal functionality, so I will basically have FDRS wait for the OnDataSent() callback from ESP-NOW. If it doesn't return "success" then FDRS will try it a few more times. 

None of this is in the official release currently, but LoRa confirmation can be found in the 'dev' branch. Any suggestions are appreciated.

timmbogner, 2022-08-13:
Alright, I now have the most current FDRS version live on the [here](https://github.com/timmbogner/Farm-Data-Relay-System/blob/two-way-comms/examples/).

I just noticed that 8266 controller compilation has an error, which I won't be able to solve until later tonight.  You can take a look or try it with an ESP32 until then.

**Quick overview:**

Basically, the gateway now has a list of peers that have subscribed to it, and I've introduced an action function sendESPNOWpeers() to send a packet to every address on the peer list.

Controllers register with their gateway on start-up, and will then be added to the gateway's peer list. The following may change as I develop: The controller has a list of 4 controller ids (in config), which correspond to four callback functions (in sketch.ino). These functions are called when data is received with a reading ID corresponding to one of the controller IDs. The callback has access to the data via a pointer passed to the function. As a warning, it is a callback function, so use care not to put much code inside of it, but also know that another may be called before the sketch returns to the loop() function.

Hope this makes sense! I'll be available if you have more questions and to keep you posted.

timmbogner, 2022-08-14:
It should be functional now. I am currently brainstorming better ways to hand data over to the user.

Maybe a set of functions: one to subscribe to a DataReading id, then another to access the latest data that's been received for that id.

garudaonekh, 2022-08-15:
Thanks, I will try it. I am now still struggling with range. i.e cannot properly work for more than 40meters if there are many trees in between. my goal is 100m on open field with tree in between because i work on farm project. My setup is Esp32-U with antenna as controller and wifi switches using esp-01.

timmbogner, 2022-08-15:
While I don't have any real proof, I've always had the impression that ESP32s had better receiving capability than 8266. If you can, try switching a ESP32 for the gateway instead. Use ESP with the external antenna as the gateway if possible.

By the way, in terms of **packet confirmation**: At this time, I think the best method for a sensor to fully confirm that a packet was successful is to **have the front-end application send a packet back** with a confirmation message using the general comm methods. The sensor will be set up as a controller as well and wait for the confirmation to arrive, resending if it doesn't.  Beyond that, FDRS can only confirm whether it reached the next gateway.

garudaonekh, 2022-08-15:
My idea is to use esp01 relay or esp12f relay because it is very cheap  and esp32 as the main controller and attach strong gain antenna to it.
esp01 relay cost 2$  on Taobao.com

timmbogner, 2022-08-15:
Just for testing's sake I'd recommend you try the ESP32 as the receiver. Perhaps even trying another ESP-01 module, as I've used many and they can sometimes be inconsistent. 

Make sure you elevate both modules as much as possible. In the newest FDRS that I linked you to, there is a pingFDRS(timeout) function for sensors/controllers, which might be helpful for range testing.

timmbogner, 2022-09-02:
Hi @garudaonekh, have you had some luck with your reception issues?
I've completed the[ relay controller sketch](https://github.com/timmbogner/Farm-Data-Relay-System/tree/dev/examples/Controller_examples/Irrigation), please let me know if there are any issues. 

As for usage, you send a DataReading out with the 'id' corresponding to the configured relay control. The type is not used, so you can set it to zero or whatever you'd like.

**For example:**
If your controller is configured as such:
```
#define CONTROL_1    133        //Address for controller 1
...
#define COIL_1       5          //Coil Pin 1
```
Then you will enable a relay connected to pin 5 of your controller when it receives the following DataReading:
```
{"id":133,"type":0,"data":1}
```

garudaonekh, 2022-09-02:
> Hi @garudaonekh, have you had some luck with your reception issues? I've completed the[ relay controller [...]

I tried ESP-07 which use external antenna. The reception is better but still not acceptable for my use cases. So I am looking into a better antenna(with reasonable cost).

timmbogner, 2022-09-02:
[Here is a resource](https://www.makeuseof.com/10-diy-long-range-wi-fi-antennas-you-can-make-at-home/) that might help. I haven't tried any of these, fyi.

closed 2022-10-02 by timmbogner