timmbogner/Farm-Data-Relay-System #111: build in Visual Studio Code I get these 4....

FMJK190, 2022-10-08:
timmbogner - Cool program you have made !!. 
When I build in Visual Studio Code I get these 4 problems/errors:

fdrs_functions.h",
	"message": "a value of type \"crcResult\" cannot be used to initialize an entity of type \"crcResult\"",
	"startLineNumber": 27,

fdrs_functions.h",
	"message": "control reaches end of non-void function [-Werror=return-type]",
	"startLineNumber": 652,

fdrs_lora.h",
	"message": "no return statement in function returning non-void [-Werror=return-type]",
	"startLineNumber": 206,

fdrs_checkConfig.h",
	"message": "missing terminating \" character",
	"startLineNumber": 61,

timmbogner, 2022-10-11:
Thanks!

I've only ever built it in Arduino.
I only recently started using VS Code as an editor, but I still don't use it to compile anything. Do you know of instructions I can follow to get to the point where i can recreate these errors? I will leave this open for a while in case someone else can shed more light on the issues.

timmbogner, 2022-10-11:
Sorry, I'm really quite new to VS Code. I haven't installed any Arduino framework on it. I'm just using it to edit the .h files in my Adruino 'libraries' folder while I compile them in Arduino IDE. Is it PlatformIO that I need?

FMJK190, 2022-10-11:
Yes, you need PlatformIO

timmbogner, 2022-10-11:
Ope, found the Arduino extension.

aviateur17, 2022-10-12:
@FMJK190, I was able to get 1_UART_Gateway from the master branch to compile correctly in PlatformIO/VSCode.

I moved all header files and ino file to src folder
Renamed .ino file to main.cpp
Added #include <Arduino.h> to top of the main.cpp file

I see some of the warnings you note but they do not prevent the project from compiling.

Are those issues preventing the compilation from completing or just letting us know about warnings?

FMJK190, 2022-10-16:
” Are those issues preventing the compilation from completing”
Yes. I have shown the errors and corrections below:

fdrs_functions.h",
"message": "a value of type "crcResult" cannot be used to initialize an entity of type "crcResult"",
"startLineNumber": 27,

enum crcResult{
  CRC_NULL,
  CRC_OK,
  CRC_BAD,
} returnCRC = CRC_NULL;

must be:
enum crcResult{
  CRC_NULL,
  CRC_OK,
  CRC_BAD
};

fdrs_functions.h",
"message": "control reaches end of non-void function [-Werror=return-type]",
"startLineNumber": 652,

    return -1;
  }
}

must be:
  }
  return -1;
}

fdrs_lora.h",
"message": "no return statement in function returning non-void [-Werror=return-type]",
"startLineNumber": 206,

  }
  return CRC_NULL;
#endif //USE_LORA
}

must be:
#endif //USE_LORA
  return CRC_NULL;
}

fdrs_checkConfig.h",
"message": "missing terminating " character",
"startLineNumber": 61,

#ifdef LOGBUF_DELAY
    DBG("log buffer delay in ms: " + " + String(LOGBUF_DELAY));

must be:
#ifdef LOGBUF_DELAY
    DBG("log buffer delay in ms: " + String(LOGBUF_DELAY));

timmbogner, 2022-10-19:
Thanks a lot! It's a busy time of the year for me, but I'll look into these fixes as soon as I get the chance.

timmbogner, 2022-10-23:
Alright, I think I got this... It did seem like it was compiling despite those problems, but the last three were definitely errors and needed correction.
@aviateur17 I fixed the other three problems and the first one (with the crcResult enum) seemed to go away? I'm assuming this is just a difference in coding practices, but this is an area of C I'm less familiar with.
Let me know if this PR fixes things on your end. I'm still learning PlatformIO.