timmbogner/Farm-Data-Relay-System #118: Serial or Mqtt comunication problem

lamasseriadipolverara, 2022-11-05:
I try to test your project as i'm a farmer and i think i can have some benefit from this.

I try the exemple come in the project, so i have:
ESPNOW_Sensor sending random reading
1_UART_Gateway reciving data. From monitor I have "Incoming ESP-NOW." 
0_MQTT_Gateway connects to wifi and mqtt server but the only mqtt send is fdrs/status = FDRS initialized

I have some errors in the output of the arduino ide 2.0

In file included from C:\Users\utente\Documents\Arduino\libraries\Farm-Data-Relay-System-main\src/fdrs_functions.h:295,
                 from C:\Users\utente\AppData\Local\Temp\.arduinoIDE-unsaved2022104-11676-x1rel0.p79zt\0_MQTT_Gateway\0_MQTT_Gateway.ino:38:
C:\Users\utente\Documents\Arduino\libraries\Farm-Data-Relay-System-main\src/fdrs_lora.h: In function 'crcResult getLoRa()':
C:\Users\utente\Documents\Arduino\libraries\Farm-Data-Relay-System-main\src/fdrs_lora.h:206:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
In file included from C:\Users\utente\AppData\Local\Temp\.arduinoIDE-unsaved2022104-11676-x1rel0.p79zt\0_MQTT_Gateway\0_MQTT_Gateway.ino:38:
C:\Users\utente\Documents\Arduino\libraries\Farm-Data-Relay-System-main\src/fdrs_functions.h: In function 'int checkPeerExpired()':
C:\Users\utente\Documents\Arduino\libraries\Farm-Data-Relay-System-main\src/fdrs_functions.h:652:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^

I try to change the pin of the serial to
#define RXD2 16
#define TXD2 17

but can't have data in mqtt server.

How can i debug that?

Sorry for my poor english

timmbogner, 2022-11-05:
Hello, thanks for trying it out!

First thing is that I believe updating to the latest version of the system should remove the warnings. Thanks for letting me know... I was aware that they were issues in PlatformIO, but I hadn't used Arduino 2.0 yet when I fixed them last week.

I checked things on my end and everything seems functional. 
From ESPNOW_Sensor
```
FDRS User Node initializing...
09:29:14.434 ->  Reading ID 1
09:29:14.434 ->  Gateway: 1
09:29:14.434 -> Initializing ESP-NOW!
09:29:14.434 ->  ESP-NOW Initialized.
09:29:14.434 -> Id: 1 - Type: 3 - Data loaded: 85.00
09:29:14.434 -> Id: 1 - Type: 1 - Data loaded: 22.07
09:29:14.434 -> Sending FDRS Packet!
09:29:14.434 -> Big Success!
09:29:14.434 -> Sleepytime!
```
From 1_UART_Gateway:
```
09:31:34.449 -> Incoming ESP-NOW.
09:31:34.449 -> Sending Serial.
09:31:34.449 -> [{"id":1,"type":3,"data":52},{"id":1,"type":1,"data":22.06900024}]
```
From 0_UART_Gateway:

```
09:35:50.370 -> Incoming Serial.
09:35:50.370 -> Sending MQTT.
```
Please update your system and let me know if you get success from the sensor node.  Ensure that you've crossed the serial lines (Rx->Tx , Tx->Rx) on your gateways too.

lamasseriadipolverara, 2022-11-08:
Thank you for the replay.

From ESPNOW_Sensor.h I have the following errors in the output during upload:

`In file included from C:\Users\utente\AppData\Local\Temp\.arduinoIDE-unsaved2022108-10300-7nd4lh.zs5wk\ESPNOW_Sensor\ESPNOW_Sensor.ino:10:
C:\Users\utente\Documents\Arduino\libraries\Farm-Data-Relay-System-main\src/fdrs_node.h: In function 'crcResult getLoRa()':
C:\Users\utente\Documents\Arduino\libraries\Farm-Data-Relay-System-main\src/fdrs_node.h:333:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
In file included from C:\Users\utente\AppData\Local\Temp\.arduinoIDE-unsaved2022108-10300-7nd4lh.zs5wk\ESPNOW_Sensor\ESPNOW_Sensor.ino:10:
C:\Users\utente\Documents\Arduino\libraries\Farm-Data-Relay-System-main\src/fdrs_node.h: In function 'bool transmitLoRa(uint16_t*, DataReading*, uint8_t)':
C:\Users\utente\Documents\Arduino\libraries\Farm-Data-Relay-System-main\src/fdrs_node.h:406:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
C:\Users\utente\Documents\Arduino\libraries\Farm-Data-Relay-System-main\src/fdrs_node.h: In function 'uint32_t pingFDRS(int)':
C:\Users\utente\Documents\Arduino\libraries\Farm-Data-Relay-System-main\src/fdrs_node.h:540:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^`

on the 1_uart_gateway
no errors in the output

on the 0_mqtt_gateway
no errors

Everythings work perfect now... thank you, sending from sensor, receving from gateway and posting in mqtt

Now it's time to go deeper and see if i can control a 4 relay board to control my irrigation system from home-assistant/node-red. 
Thank you very much

timmbogner, 2022-11-09:
Awesome! I just found my own 4 channel relay board, and updated the code for it. In case you don't know, my [Irrigation sketch is actually a 4-channel relay driver](https://github.com/timmbogner/Farm-Data-Relay-System/blob/main/examples/Controller_examples/Irrigation/Irrigation.ino).

To use it, set the relay pins in the .ino file, then flash it to your board. After powering up the relay module, enter the following into the serial monitor of your gateway: ```[{"id":101,"type":1,"data":1}]```. This should trigger the first relay. To do it from MQTT will involve publishing that same JSON string to the topic "fdrs/command".

lamasseriadipolverara, 2022-11-11:
Everithings work correctly... but
When i connect the ESP32s (mqttgatway and espnow gateway) to a hi-link stepdown 230 -> 5 volts seems that serial connection doesn't work.

Test that i do:
connect all the esp from usb -> mqtt connected / serial works / irrigation receive 101 from mqtt->espnow 
connect mqttgw and espnowgw to hi-link, irrigation to usb with serial monitor running -> esp-now peer registration accepted, mqtt inizialized (from mqtt explorer), send mqtt [{"id":101,"type":1,"data":1}] nothing in the serial monitor of irrigation...
Hope i explain it.
Do you have same idea?

timmbogner, 2022-11-11:
Thanks for trying the controller sketch! You're one of the first to do so.

Tell me a little more about your "mqttgw and espnowgw" circuit, especially when it is attached to the hi-link. Both devices are connected to the same ground, correct?

The next thing to find out is whether the "mqttgw" is outputting serial correctly. Can you monitor the serial port while it is running off the hi-link?

OH, and an afterthought: MQTT gateways send a status message at boot. Look for the message "FDRS initialized" on the MQTT topic *fdrs/status*. That would confirm it is functioning, which is handy.

lamasseriadipolverara, 2022-11-12:
It's not the exact part i used (it's the first time i use fritzing) but you can have the idea of the connection

[image]

`Tell me a little more about your "mqttgw and espnowgw" circuit, especially when it is attached to the hi-link. Both devices are connected to the same ground, correct?`

No each esp32 is connect to one hi link stepdown...

`The next thing to find out is whether the "mqttgw" is outputting serial correctly. Can you monitor the serial port while it is running off the hi-link?`

yes, when attach to usb everythings wark correctly mqttgw take mqtt input and pass it to serial, espnowgw take the serial message and pass it throw espnow, irrigation serial monitor get 101 and open the relay.
So, when all the esp is connected to usb the system work smoothly

`OH, and an afterthought: MQTT gateways send a status message at boot. Look for the message "FDRS initialized" on the MQTT topic fdrs/status. That would confirm it is functioning, which is handy.`
Yes, in fdrs/status i have the "inizialized" message both connected to usb and to hi-link, but when connected to hi link no message pass throw.

jcoacro, 2022-11-12:
@lamasseriadipolverara Since you are powering them from different psu, try connecting the grounds from the psu together.

timmbogner, 2022-11-12:
@jcoacro is correct. The two modules need to have a common ground connection. Either run a wire between the two controller grounds or the power supply grounds.

I'm not familiar with your HiLink seltepdown module, but it seems like you should be able to use one to power them both.

lamasseriadipolverara, 2022-11-12:
I don't understand if i have to:
1) connect both esp ground to 1 of the hi-link
or
2) connect togher the esp ground pin (and connect each esp to each hi-link ground)
or
3) connect the 2 ground of hi link, then connect both esp ground to the hi link

But as i would like to understand... why i have to connect the ground togher?

Next week i'm out for work, when i come back i try your suggestions. 
Thankyou very much, you are awesome!!!

timmbogner, 2022-11-12:
Sorry I was confusing. Both 2 and 3 are correct, and are essentially the same thing.

As for why... it is generally essential for any two devices that are communicating by wire to have the same ground. Sorry to give you a vague "it's just the way it is" answer, but I don't have the background to [explain it properly](https://forum.arduino.cc/t/common-ground-and-why-you-need-one/626215/2).

[image]

lamasseriadipolverara, 2022-11-21:
Perfect, everything works.
Thank you

closed 2022-11-22 by timmbogner