timmbogner/Farm-Data-Relay-System #126: PM data types

Stips5, 2022-12-28:
Hi Tim

I think PM1, PM2.5, PM10 (particulate matter) could be added to datatypes list

Stips5, 2022-12-28:
Suggested [change](https://github.com/Stips5/Farm-Data-Relay-System/commit/353d426112232e3ab1bdb1e1a0c3bb1a6ddff0ed)

timmbogner, 2023-01-03:
Hey! I'm just getting back to things. This will be great, although to stay in line with the [type standard](https://github.com/timmbogner/Farm-Data-Relay-System/blob/main/src/fdrs_datatypes.h) I've been using it should go as follows:

```
#define PM1_T	27 // 1 Particles
#define PM2_5_T	28 // 2.5 Particles
#define PM10_T	29 // 10 Particles
```
If you're good making a pull request, I'll accept it. Otherwise I can do it and give you credit.
What sensor are you using?

Stips5, 2023-01-03:
I can try with pull request, also I would add UV_T representing UV value, eg. using Adafruit_VEML6070
I am using PMS5003 for reading PM values.

Stips5, 2023-01-07:
Merged with https://github.com/timmbogner/Farm-Data-Relay-System/commit/a597da242bfdb8b9bf1dc24573b3e180cb398d9a