timmbogner/Farm-Data-Relay-System #132: compile error

pawelchwast, 2023-02-15:
I can not compile any sensor examples.
Always same error:
fdrs_node.h: In function 'crcResult getLoRa()'
error: no return statement in function returning non-void

I have also try compile upcoming pull request version with same results.
Any suggestions ?

timmbogner, 2023-02-15:
Are you in PlatformIO or Arduino?
Could you tell me if [this branch](https://github.com/timmbogner/Farm-Data-Relay-System/tree/fix_132) fixes it for you?

pawelchwast, 2023-02-16:
I am using Arduino ide version 2.
On this branch, i have other error:
fdrs_node.h:785:1: error: control reaches end of non-void function [-Werror=return-type]
i was try to compile capacitive soil example sensor and esp now sensor.
both gives same error

timmbogner, 2023-02-16:
This is a slight mystery. The problem you're describing is an error known on PlatformIO, but I've never had it occur on Arduino. Could you tell me what OS and board definition that you are using? I can't seem to replicate the problem in Windows 10. 

It sounds like reestablishing PlatformIO compatibility will fix this either way. I am overdue to work on that, so I'll try to get to it today.

timmbogner, 2023-02-16:
There were less issues than I anticipated, so I think the changes I just made to the [fix_132 branch](https://github.com/timmbogner/Farm-Data-Relay-System/tree/fix_132) should fix your problem.

pawelchwast, 2023-02-17:
I can confirm, that capacitive soil sensor example is now compiling without any issue :)
Many thanks :)
I'm on mac ventura with intel cpu and NodeMCU board
Next i will try to configure and compile esp-now to lora gateway with ebyte e32 433Mhz board. 
I don't known yet, how to configure radio lib to use it, so it will be fun.
best regards
Paul.

pawelchwast, 2023-02-17:
I had found that radio lib does not support uart (witch j have in this module) but only spi.
hmm

timmbogner, 2023-02-17:
> I can confirm, that capacitive soil sensor example is now compiling without any issue :) Many thanks :) I'm on mac ventura with intel cpu and NodeMCU board

Aha, there must be some difference between Mac and Windows Arduino compilation. I'll have to keep a closer eye on that in the future!

> I had found that radio lib does not support uart (witch i have in this module) but only spi.

Yes. FDRS needs to control the LoRa radio directly, whereas these UART-based LoRa modules add an extra MCU to interface UART and the radio chip as well as add some addressing. They are handy on their own, but (as far as I know) aren't compatible with other LoRa hardware.