timmbogner/Farm-Data-Relay-System #133: new data types: power and energy

generalmanager, 2023-02-22:
I've got a smart electricity meter which can output the current power usage in watts as well as the daily/weekly/yearly energy usage in watthours, that I'd like to integrate into fdrs, so it'd be great to have matching data types. Those could also come in very handy when monitoring a remote battery backed solar installation.

timmbogner, 2023-02-22:
I'll put POWER_T and POWER2_T on the new release for sure. These will match the  VOLTAGE_T & VOLTAGE2_T and CURRENT_T & CURRENT2_T which are doubled to allow you to monitor your own voltage, plus a second device like a battery or solar panel.

I'm going to take some time to think about how to treat daily/weekly/yearly total. I currently am thinking they will be defined by one type, and you'll need to send them each with a different ID. In the meantime, just use an arbitrary value (30-255 should be okay) for the type.

generalmanager, 2023-02-22:
Thanks a bunch Timm! I agree that it should probably just be ENERGY_T and maybe ENERGY2_T as there are way too many totals to cover everybodies usecases. Using IDs seems more sensible as it is a lot more flexible. This would also keep everything nicely aligned with the other types. :+1:

timmbogner, 2023-02-23:
Got this added. Sorry I forgot to list you as a co-author, which I'd normally do. I'll get you on one related to ethernet. I'm going to tackle that soon-ish.

closed 2023-02-23 by timmbogner