timmbogner/Farm-Data-Relay-System #138: Verifying ESPNOW_Sensor

Jzee01, 2023-02-25:
I am trying to upload the ESPNOW_Sersor file to an esp8266. I keep getting these errors. 

In file included from C:\Users\jzueh\AppData\Local\Temp\.arduinoIDE-unsaved2023125-13764-1r1er1j.fll2h\ESPNOW_Sensor\ESPNOW_Sensor.ino:10:
c:\Users\jzueh\Documents\Arduino\libraries\Farm-Data-Relay-System-main\src/fdrs_node.h: In function 'crcResult getLoRa()':
c:\Users\jzueh\Documents\Arduino\libraries\Farm-Data-Relay-System-main\src/fdrs_node.h:452:1: error: no return statement in function returning non-void [-Werror=return-type]
  452 | }
      | ^
c:\Users\jzueh\Documents\Arduino\libraries\Farm-Data-Relay-System-main\src/fdrs_node.h: In function 'uint32_t pingFDRS(int)':
c:\Users\jzueh\Documents\Arduino\libraries\Farm-Data-Relay-System-main\src/fdrs_node.h:599:1: error: control reaches end of non-void function [-Werror=return-type]
  599 | }
      | ^
cc1plus.exe: some warnings being treated as errors

exit status 1

Compilation error: exit status 1

I am new to this so I'm sure if i did something wrong or need to change something.

timmbogner, 2023-02-25:
No, you aren't in the wrong. This is an issue that involves differences between Arduino versions (or something, not sure to be honest).
It should be fixed in the latest version,[ which you can find here. ](https://github.com/timmbogner/Farm-Data-Relay-System/tree/release_2023_01)
What exact system are you using Arduino under? I knew this had been an issue under Mac, but that looks like Windows.

Jzee01, 2023-02-26:
I switched to the latest version and it worked. Thank you.

Rikolas, 2023-02-27:
Hi Timm
trying to get it work. I am on windows, Arduino 2.0.3. Only using ESP-now (for now-no lora)
I had the same issue, with the new version I get lots of errors in fdrs_node_lora.h. 
Do you have a quick fix? 
Thanks and continue your awesome work. 

In file included from c:\xxx\ArduinoIDE\libraries\Farm-Data-Relay-System-main\src/fdrs_node.h:66,
                 from C:\Uxxx\ArduinoIDE\libraries\Farm-Data-Relay-System-main\examples\ESPNOW_Sensor\ESPNOW_Sensor.ino:10:
c:\xxx\ArduinoIDE\libraries\Farm-Data-Relay-System-main\src/fdrs_node_lora.h: In function 'crcResult handleLoRa()':
c:\xxx\ArduinoIDE\libraries\Farm-Data-Relay-System-main\src/fdrs_node_lora.h:143:1: error: no return statement in function returning non-void [-Werror=return-type]
  143 | }
      | ^
In file included from c:\xxx\ArduinoIDE\libraries\Farm-Data-Relay-System-main\src/fdrs_node.h:66,
                 from C:\xxx\ArduinoIDE\libraries\Farm-Data-Relay-System-main\examples\ESPNOW_Sensor\ESPNOW_Sensor.ino:10:
c:\xxx\ArduinoIDE\libraries\Farm-Data-Relay-System-main\src/fdrs_node_lora.h: In function 'crcResult transmitLoRa(uint16_t*, DataReading*, uint8_t)':
c:\xxx\ArduinoIDE\libraries\Farm-Data-Relay-System-main\src/fdrs_node_lora.h:281:1: error: no return statement in function returning non-void [-Werror=return-type]
  281 | }
      | ^
c:\xxx\ArduinoIDE\libraries\Farm-Data-Relay-System-main\src/fdrs_node_lora.h: In function 'crcResult transmitLoRa(uint16_t*, SystemPacket*, uint8_t)':
c:\xxx\ArduinoIDE\libraries\Farm-Data-Relay-System-main\src/fdrs_node_lora.h:326:1: error: no return statement in function returning non-void [-Werror=return-type]
  326 | }
      | ^
c:\xxx\ArduinoIDE\libraries\Farm-Data-Relay-System-main\src/fdrs_node_lora.h: In function 'crcResult getLoRa()':
c:\xxx\ArduinoIDE\libraries\Farm-Data-Relay-System-main\src/fdrs_node_lora.h:485:1: error: no return statement in function returning non-void [-Werror=return-type]
  485 | }
      | ^
c:\xxx\ArduinoIDE\libraries\Farm-Data-Relay-System-main\src/fdrs_node_lora.h: In function 'uint32_t pingFDRSLoRa(uint16_t*, uint32_t)':
c:\xxx\ArduinoIDE\libraries\Farm-Data-Relay-System-main\src/fdrs_node_lora.h:513:1: error: no return statement in function returning non-void [-Werror=return-type]
  513 | }
      | ^
cc1plus.exe: some warnings being treated as errors

exit status 1

Compilation error: exit status 1

timmbogner, 2023-02-27:
Thanks @Rikolas, this is (again) Arduino seemingly compiling differently for different systems. At any rate, I think I fixed it on this branch. Let me know how it works.

@aviateur17 Apparently sometimes Arduino doesn't like when all return statements are protected by macros, so keep an eye on that.

timmbogner, 2023-02-27:
@Rikolas [forgot the link](https://github.com/timmbogner/Farm-Data-Relay-System/tree/fix_138)

aviateur17, 2023-02-27:
> @aviateur17 Apparently sometimes Arduino doesn't like when all return statements are protected by macros, so keep an eye on that.

Sounds good.  I'm pretty sure I tested against Arduino 1.8.19 but maybe something is different in the 2.0 branch of Arduino or maybe there is some configuration that is different.  I'll go back and look at the change you made and do some testing.  Thanks for the heads up.

timmbogner, 2023-02-27:
I tested it on 2.0.3 yesterday on my Win10 machine and it was fine. I've known that PlatformIO acts this way, but I've only recently been getting reports of users seeing it in Arduino.

timmbogner, 2023-02-27:
@Rikolas I'm so sorry, the fix didn't upload before my last post. but It's there now.

I've gotta get my coffee *before* I check my emails.

aviateur17, 2023-02-27:
@timmbogner, I see the changes you made in that branch_138.  Odd that there were no errors when I tested compilation in Arduino.  Since you know how to reproduce can you tell me what is the difference in how Rikolas has his setup so I can easily reproduce the error?   Hopefully this silliness will go away once we get those ifdefs out of the header files with all the functions and around the #include of certain header files.

timmbogner, 2023-02-27:
I think the errors will always occur or at least show as warnings in PIO. I can't reproduce them in Arduino on my Windows machine, I'll try Ubuntu later. Someone reported the issue on Mac recently in #132 , which is when I fixed some pre-existing occurrences of the problem.

aviateur17, 2023-02-27:
Okay thanks.  I use Linux Mint (based off of Ubuntu) for my main development platform but I also have a MAC so I'll see if I can reproduce this on my MAC.

Rikolas, 2023-02-27:
Working! Thank you.

closed 2023-03-02 by timmbogner