timmbogner/Farm-Data-Relay-System #145: Sensor crashing stack overflow on ESP8266 12F board

machasm1, 2023-03-02:
Getting the error below when uploading code to ESP8266 12F board.
ESP Decoder does not function under Arduino IDE 2.0 so struggling to find the root cause of the problem.
Any ideas?

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffd60 end: 3fffffd0 offset: 01a0
3fffff00:  00000000 00000000 00000020 40100188  
3fffff10:  00000014 00000000 00000000 40101e98  
3fffff20:  3ffe8fd0 40231d4f 3ffec8a8 3ffef8c4  
3fffff30:  00000000 0031d888 3ffee644 00000000  
3fffff40:  3ffe85c8 0000000e 00000020 00000000  
3fffff50:  3ffee644 00000000 00000cc2 40202971  
3fffff60:  00000000 00000cc2 b126e978 000ec11c  
3fffff70:  00000015 3ffefd14 3ffee649 3ffeeff0  
3fffff80:  3ffee644 00000cc2 00000000 40203228  
3fffff90:  40201a10 3ffee648 00000000 40201285  
3fffffa0:  3fffdad0 00000000 3ffeefc4 402014c7  
3fffffb0:  3fffdad0 00000000 3ffeefc4 40202a1c  
3fffffc0:  feefeffe feefeffe 3fffdab0 40100dc5  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v000440c0
~ld
FDRS User Node initializing...
 Reading ID 1
 Gateway: 2
Initializing ESP-NOW!
 ESP-NOW Initialized.
Id: 1 - Type: 3 - Data loaded: 66.00
Id: 1 - Type: 1 - Data loaded: 22.07
Sending FDRS Packet!

timmbogner, 2023-03-02:
To be clear: Does the error occur after the ESP-NOW message was sent? What board definition are you compiling under?

Could you try it on[ this newer version](https://github.com/timmbogner/Farm-Data-Relay-System/tree/release_2023_01) and tell me if the error is still present?

machasm1, 2023-03-02:
Hi Thanks for the quick response.
It just keeps looping round that whole error message as described above.
This is with the newest release that you pointed me to (23_01).
I haven't tried it on the release version.
I am using the Arduino IDE 2 and the board is a NodeMCU 1.0 (ESP12E Module).

Getting a compilation error when trying to use the stable release version as shown

In file included from C:\Users\joe\Documents\Arduino\libraries\Farm-Data-Relay-System-main\examples\ESPNOW_Sensor\ESPNOW_Sensor.ino:10:
c:\Users\joe\Documents\Arduino\libraries\Farm-Data-Relay-System-main\src/fdrs_node.h: In function 'crcResult getLoRa()':
c:\Users\joe\Documents\Arduino\libraries\Farm-Data-Relay-System-main\src/fdrs_node.h:452:1: error: no return statement in function returning non-void [-Werror=return-type]
  452 | }
      | ^
c:\Users\joe\Documents\Arduino\libraries\Farm-Data-Relay-System-main\src/fdrs_node.h: In function 'uint32_t pingFDRS(int)':
c:\Users\joe\Documents\Arduino\libraries\Farm-Data-Relay-System-main\src/fdrs_node.h:599:1: error: control reaches end of non-void function [-Werror=return-type]
  599 | }
      | ^
cc1plus.exe: some warnings being treated as errors
Multiple libraries were found for "fdrs_globals.h"
  Used: C:\Users\joe\Documents\Arduino\libraries\Farm-Data-Relay-System-main
  Not used: C:\Users\joe\Documents\Arduino\libraries\Farm-Data-Relay-System-new
exit status 1

Compilation error: exit status 1

timmbogner, 2023-03-02:
I'll check some things and get back to you soon. But in the meantime: make sure that you don't have two copies of FDRS in your 'libraries' folder.  It looks like you installed the new version with a different folder name, but this will confuse Arduino. Delete "C:\Users\joe\Documents\Arduino\libraries\Farm-Data-Relay-System-main" and try again.

machasm1, 2023-03-02:
Yeah already noticed the duplicate folder problem so took the main one out of the libraries folder.
As as side note I am trying to run this on a D1 mini esp8266 if that makes any difference.
Also same problem occurs when trying to run the LoRa sensor.

aviateur17, 2023-03-02:
Wonder if it's getting stuck in this loop:  fdrs_node.h line 469:

`while(esp_now_ack_flag == CRC_NULL){
    delay(0);
  }`

There is no yield() so if the loop does not return the controller watchdog timer will expire and reset the CPU.
probably best to add a yield(); command there if there isn't one in the current branch.

So then the question is why no ack flag.   I haven't spent much time going through the scenarios but do you have a ESP-NOW gateway configured and listening at address 2?

aviateur17, 2023-03-02:
> Also same problem occurs when trying to run the LoRa sensor.
Can you send the debug output for the LoRa sensor sketch?

timmbogner, 2023-03-02:
Good save, Jeff! It looks like I was under the impression that delay(0) would also yield(), and maybe it does in ESP32. 

I'm wondering if the LoRa issue isn't the same thing in a different spot.

timmbogner, 2023-03-02:
@machasm1 Try the new branch again and let me know if ESP-NOW works better.

machasm1, 2023-03-02:
WooHoo Well done guys.
Problem solved and in double time too.

timmbogner, 2023-03-02:
Big thanks to both of you!

On LoRa the issue is ```while (true);``` after RadioLib init fails. On ESP32 it just hangs, but on ESP8266 it crashes. There's probably a better way to do that.

closed 2023-03-03 by timmbogner