timmbogner/Farm-Data-Relay-System #147: New Update

timmbogner, 2023-03-04:
I have updated the main branch with the newest version of FDRS. The [node ](https://github.com/timmbogner/Farm-Data-Relay-System/blob/main/extras/Node.md )and [gateway](https://github.com/timmbogner/Farm-Data-Relay-System/blob/main/extras/Gateway.md) documentation should reflect all new additions and changes made to the API.

The new version is fully compatible with the previous one... it just works better 😎

 Please let me know right away if you encounter any bugs or feedback. 
 
 **Thank you to everyone who has helped!**

aviateur17, 2023-03-06:
@timmbogner, did you want to start a new devel or test branch to post new features/bug fixes against and eventually to be pulled into main after some time?

timmbogner, 2023-03-06:
Maybe... I'd almost prefer to just have branches for specific lines of development. Like a timekeeping branch, cellular, etc. 
What would you prefer?

aviateur17, 2023-03-06:
Either way.  Just wondering if you would roll feature A, feature B, feature C into a testing or devel branch and then wait a month or so and roll all of those into main.   But if you want to roll Feature A into main and Feature B into main and Feature C into main that works too.  Just wondering if I should develop against main branch or develop against a testing or devel branch.

timmbogner, 2023-03-06:
I'd say go against the main branch. We'll just orchestrate  I like keeping experiments separate, but when they stray from the main branch and I then have to finish the new branch to fix bugs in the main I get stressed.  It's a trap I keep falling into, but I figure I'll get the hang of it. I'm def better than last summer, which was a conflict-resolving nightmare for me 😅

Speaking of, I will go through at some point and try to resolve the all conflicts I've caused on your Timekeeping branch.

timmbogner, 2023-03-06:
In my ideal situation, a new release would be comprised of me just pulling 3-4 finalized pull requests.

closed 2023-03-25 by timmbogner