timmbogner/Farm-Data-Relay-System #150: Esp8266 freeze

Stips5, 2023-03-06:
Is anyone havig problem with esp8266 freezing after 10-15 minutes, I have following hierarchy.
Esp8266 with lora and mqtt which freezes after 15minutes, 
It receives lora msgs from esp32, so far no problems with it.
Esp receives msg from another esp8266 through espnow and it also freezes after 15 minutes.
So my thought is that it has to be some memory or cpu issue. Since its working same when connected to pc and when connected to power grid.
No error msg are being printed in console. Any ideas?

timmbogner, 2023-03-07:
I'm running a "torture test" sending data from an ESP32 ESP-NOW "stress test" sensor to a Wemos8266 gateway sending via ESP-NOW to ESP32 gateway, linked via UART to Wemos8266 MQTT gateway. They've been buzzing along for 30 minutes without issue.

I'll let this run for a while and get back to you tomorrow. What ESP8266 module are you using? Are you using the latest version of FDRS?

Stips5, 2023-03-07:
I am running version where pm data types where added, I am using nodemcu esp8266's.
My sensor sends data every 10 seconds, at what rate do you send data?

```
  sendFDRS();
  sleepFDRS(10);  //Sleep time in seconds
```

timmbogner, 2023-03-07:
The stress test sends every 15 milliseconds, and it's all still going on mine.

Try updating at least the 8266 devices to the latest FDRS version and see how they work. I made a lot of changes (behind the scenes) since your version, maybe something will have fixed it.

Stips5, 2023-03-07:
```
I've also added this to my MQTT gateway but its not reason why sensor fails

void mqtt_publish(const char* payload) {
#ifdef USE_WIFI

  //TODO extract to config paramet
```er
  client.setBufferSize(512);

Ill try with updateing FDRS

Stips5, 2023-03-07:
Yes, adding #define USE_SX1278 worked, now it works with 
```
#define USE_SX1278
#define LORA_FREQUENCY 433.0
#define RADIOLIB_MODULE SX1278  // ESP8266 SX1278
```

I'am heading to test freeze.

Stips5, 2023-03-08:
After switching to newest version it works like a charm, stress tested it for 12h, no problems so far, thanks mate :)

closed 2023-03-08 by Stips5

timmbogner, 2023-03-08:
Excellent!

By the way, ```USE_SX1278``` isn't actually needed. ```USE_SX126X``` is the only version of that option, if that's not set then the system will default to using SX127x methods.