timmbogner/Farm-Data-Relay-System #154: MQTT

Srgadget, 2023-03-20:


Srgadget, 2023-03-20:
Hello Sir I first wanted to say many thanks for the project and all the work you did and sharing you r knowledge  and lesson's learned. my question is I have my MQTT broker and Node-Red sitting on Home Assistant. Rather than using a serial connection can we send the data straight from Lora Gateway to my MQTT via wifi ? many thanks again ... and If I opt to use this I will gladly make a donation

nglessner, 2023-03-20:
Yep, that will work. Or you can literally use a USB cable to power the
gateway and plug that straight into your HA instance. Use serial in Node
red to consume the data. This is what I do, to cut out that wifi hop.

timmbogner, 2023-03-21:
Hello! Thanks!
Yep, you are free to use WiFi and LoRa on the same gateway. The big limitation is that you can't use ESP-NOW and WiFi,

timmbogner, 2023-03-21:
Awesome! Let me know if you have any problems. I have recently upgraded the system's LoRa capabilities and am still working [through some bugs](https://github.com/timmbogner/Farm-Data-Relay-System/issues/155).

closed 2023-03-25 by timmbogner