timmbogner/Farm-Data-Relay-System #156: LORA and ESPNOW don't work togher

lamasseriadipolverara, 2023-03-26:
i have this configuration:
LORA receving/ESPnow sending in 0x02
ESPNow irrigation controller in 0x03

If i have both LORA and ESPNow configured in 0x02, 0x03 can't register to the gateway. If I have only ESPNow configured in 0x02, 0x03 correctly register to the gateway.

This is the configuration:
0x02 (as you see i commented the //#define USE_LORA and in this way espnow registration of 0x03 in the gateway was successfully)
//  FARM DATA RELAY SYSTEM
//
//  GATEWAY 2.000 Configuration

#include <fdrs_globals.h>
#define FDRS_DEBUG     //Enable USB-Serial debugging
#define DEBUG_CONFIG // Displays full config info on startup

#define UNIT_MAC     0x02  // The address of this gateway

// Actions -- Define what happens when a packet arrives at each interface:
// Current function options are: sendESPNow(MAC), sendSerial(), sendMQTT(),
//  sendLoRaNbr(interface) , sendESPNowNbr(interface).

#define ESPNOWG_ACT    sendLoRaNbr(1);
#define SERIAL_ACT         
#define MQTT_ACT          
#define LORAG_ACT      sendLoRaNbr(1),sendESPNow(0x03) ; 

// Protocols -- Define which protocols the gateway will use.
// Warning: ESP-NOW and WiFi should not be used simultaneously.

#define USE_ESPNOW  
//#define USE_LORA
//#define USE_WIFI    //Used only for MQTT gateway

// Neighbor addresses
#define ESPNOW_NEIGHBOR_1 0x00  // ESPNOW1 Address 
#define ESPNOW_NEIGHBOR_2 0x00  // ESPNOW2 Address
#define LORA_NEIGHBOR_1    0x01  // LoRa1 Address
#define LORA_NEIGHBOR_2    0x04  // LoRa2 Address

// Neighbor Actions
#define ESPNOW1_ACT   
#define ESPNOW2_ACT                    
#define LORA1_ACT  broadcastLoRa();
#define LORA2_ACT  sendLoRaNbr(1); 

// LoRa Configuration
// LoRa Configuration
#define RADIOLIB_MODULE SX1276 //Tested on SX1276
//#define USE_SX126X
#define LORA_SS  18
#define LORA_RST 19
#define LORA_DIO 26
#define LORA_TXPWR 17   // LoRa TX power in dBm (: +2dBm - +17dBm (for SX1276-7) +20dBm (for SX1278))
#define LORA_ACK        // Request LoRa acknowledgment.

#define CUSTOM_SPI
#define LORA_SPI_SCK 32
#define LORA_SPI_MISO 19
#define LORA_SPI_MOSI 27
//Pins for UART data interface (ESP32 only)
#define RXD2 14
#define TXD2 15

// WiFi and MQTT Credentials  -- These will override the global settings
//#define WIFI_SSID   "Your SSID"  
//#define WIFI_PASS   "Your Password"

//#define MQTT_ADDR   "192.168.0.8"
//#define MQTT_PORT   1883 // Default MQTT port is 1883
//#define MQTT_AUTH   //Enable MQTT authentication 
//#define MQTT_USER   "Your MQTT Username"
//#define MQTT_PASS   "Your MQTT Password"

//Logging settings  --  Logging will occur when MQTT is disconnected
//#define USE_SD_LOG        //Enable SD-card logging
//#define USE_FS_LOG        //Enable filesystem (flash) logging
#define LOGBUF_DELAY 10000  // Log Buffer Delay - in milliseconds
#define SD_SS 0             //SD card CS pin (Use different pins for LoRa and SD)
#define LOG_FILENAME "fdrs_log.csv"

///#define USE_OLED
#define OLED_HEADER "FDRS"
#define OLED_SDA 4
#define OLED_SCL 15
#define OLED_RST 16

This is for 0x03
//  FARM DATA RELAY SYSTEM
//
//  Sensor Configuration

#include <fdrs_globals.h>

#define READING_ID    1   //Unique ID for this sensor
#define GTWY_MAC      0x02 //Address of the nearest gateway

#define USE_ESPNOW
//#define USE_LORA
//#define DEEP_SLEEP
//#define POWER_CTRL    14
#define FDRS_DEBUG

// LoRa Configuration
#define RADIOLIB_MODULE SX1276 //Tested on SX1276
#define LORA_SS 18
#define LORA_RST 14
#define LORA_DIO 26

#define LORA_TXPWR 17    // LoRa TX power in dBm (: +2dBm - +17dBm (for SX1276-7) +20dBm (for SX1278))
#define LORA_ACK        // Request LoRa acknowledgment.

With this configuration i have in the serial monitor:
 Reading ID 1
 Gateway: 2
Initializing ESP-NOW!
 ESP-NOW Initialized.
ESP-NOW peer registration request submitted to 2
Registration accepted. Timeout: 300000
Subscribing to DataReading ID 101
Subscribing to DataReading ID 102
Subscribing to DataReading ID 103
Subscribing to DataReading ID 104
FARM DATA RELAY SYSTEM :: Irrigation Module

PS by the way i commented //bool newData = false; in irrigation.ino because if not, the sketch did'nt compile correctly...

timmbogner, 2023-03-26:
Hello, 
First, the Irrigation sketch had an error with ```newData```. I'm not sure how I didn't notice it initially, but it should be fixed now.  Thank you for bringing it to my attention!

At first I was indeed able to recreate your error, as my pings and registration requests went unanswered. Then I switched to my gateway's serial console and realized that due to a LoRa initialization problem my gateway was not starting up. I had selected the wrong board, which caused a RadioLib error -2. After fixing it, the gateway initialized correctly and everything worked as intended.

Is your gateway 0x02 definitely booting successfully? It looks like you're using an older configuration page that doesn't include the LORA_BUSY definition, which is now required if using the latest version. You can just add ```#define LORA_BUSY  RADIOLIB_NC```  to fix that.

timmbogner, 2023-03-26:
You also haven't called ```sendESPNowPeers()``` anywhere, so devices can connect, but won't ever get any data.

Just for testing's sake: change ```#define SERIAL_ACT``` to ```#define SERIAL_ACT  sendESPNowPeers();```

Now you can send commands directly to the irrigation controller by entering them into the console and sending. For example to set the first relay enter this:

```[{ id: 101, type: 0, data: 1 }]```

lamasseriadipolverara, 2023-03-27:
Ok now the registration request is ok. But espnow irrigation didn't receive any message. Be careful for my system organization i have change the unit_mac now lora->espnow is 0x02.
This is the configuration file and the serial:
For 0x02 lora->EspNow
//  FARM DATA RELAY SYSTEM
//
//  GATEWAY CONFIGURATION

//Addresses
#define UNIT_MAC           0x02  // The address of this gateway

#define ESPNOW_NEIGHBOR_1  0x03  // Address of ESP-NOW neighbor #1
#define ESPNOW_NEIGHBOR_2  0x00  // Address of ESP-NOW neighbor #2
#define LORA_NEIGHBOR_1    0x01  // Address of LoRa neighbor #1
#define LORA_NEIGHBOR_2    0x05  // Address of LoRa neighbor #2

// Interfaces
#define USE_ESPNOW  
#define USE_LORA
//#define USE_WIFI  // Will cause errors if used with ESP-NOW. Use a serial link instead!
//#define USE_ETHERNET

// Routing
// Options: sendESPNowNbr(1 or 2); sendESPNowPeers(); sendLoRaNbr(1 or 2); broadcastLoRa(); sendSerial(); sendMQTT();
#define ESPNOWG_ACT    
#define LORAG_ACT      sendLoRaNbr(1); sendESPNow(1);
#define SERIAL_ACT     sendESPNowPeers();
#define MQTT_ACT          
#define INTERNAL_ACT   sendLoRaNbr(1);
#define ESPNOW1_ACT    
#define ESPNOW2_ACT                    
#define LORA1_ACT      sendLoRaNbr(2); broadcastLoRa(); sendESPNowPeers();
#define LORA2_ACT      sendLoRaNbr(1);

// LoRa Configuration
#define RADIOLIB_MODULE SX1276
#define LORA_SS    18
#define LORA_RST   21
#define LORA_DIO   26
#define LORA_BUSY  RADIOLIB_NC
//#define USE_SX126X

#define LORA_TXPWR 17   // LoRa TX power in dBm (: +2dBm - +17dBm (for SX1276-7) +20dBm (for SX1278))

#define CUSTOM_SPI
#define LORA_SPI_SCK  32
#define LORA_SPI_MISO 19
#define LORA_SPI_MOSI 27

#define FDRS_DEBUG     // Enable USB-Serial debugging

serial monitor:
Address:2
10:04:05.308 -> RadioLib initialization successful!
10:04:05.308 -> LoRa Initialized. Frequency: 868.00  Bandwidth: 125.00  SF: 7  CR: 5  SyncWord: 18  Tx Power: 17dBm
10:04:05.308 -> Initializing ESP-NOW!
10:04:05.404 ->  ESP-NOW Initialized.
10:04:14.762 -> ESP-NOW System Packet
10:04:14.762 -> Ping back to sender
10:04:14.762 -> ESP-NOW System Packet
10:04:14.762 -> Device requesting peer registration
10:04:37.328 -> Incoming Serial.
10:04:37.328 -> Sending to ESP-NOW peers.

For EspNOW irrigation module:
//  FARM DATA RELAY SYSTEM
//
//  Sensor Configuration

#include <fdrs_globals.h>

#define READING_ID    1   //Unique ID for this sensor
#define GTWY_MAC      0x02 //Address of the nearest gateway

#define USE_ESPNOW
//#define USE_LORA
//#define DEEP_SLEEP
//#define POWER_CTRL    14
#define FDRS_DEBUG

serial monitor:
 ESP-NOW Initialized.
 ESP-NOW ping sent.
ESP-NOW Ping Reply in 3ms from aa:bb:cc:dd:ee:2
ESP-NOW peer registration request submitted to 2
Registration accepted. Timeout: 300000
Subscribing to DataReading ID 101
Subscribing to DataReading ID 102
Subscribing to DataReading ID 103
Subscribing to DataReading ID 104
FARM DATA RELAY SYSTEM :: Irrigation Module

timmbogner, 2023-03-27:
I apologize, this looks like I accidentally deleted a line the other day when editing Irrigation.ino.
Github uploading seems to be down currently, but to fix it add ```loopFDRS();``` at the beginning of the loop function.

timmbogner, 2023-03-27:
@lamasseriadipolverara There seems to be some further problem with ESP-NOW controllers. I'll let you know what I figure out.

lamasseriadipolverara, 2023-03-28:
I'm working with esp32, and espnow now seems to works correctly when i used the serial monitor in 0x02.
But when in 0x02 i try to pass throw lora to espnow lora doesn't work. If i have only Lora configureted in 0x02 i receved the message from the gateway, when i enable espnow i didn't have any message....

timmbogner, 2023-03-28:
I can't seem to recreate the problem. Tell me if I'm understanding correctly:
You are trying to get data from Gateway 0x01 to Gateway 0x02 via LoRa, then from Gateway 0x02 to a Controller via ESP-NOW. When you send serial data to Gateway 0x02, it successfully sends to Controller. However, when you send the data from Gateway 0x01 to Gateway 0x02 via LoRa, Gateway 0x02 does not send the data to Controller via ESP-NOW.

I copied your configs into a sketch and changed to my own LoRa configs, then I used the 1_UART_Gateway sketch and modified the LoRa neighbor 2 to 0x02, and enabled USE_LORA. 
With this setup I can send data to the serial monitor of Gateway 0x01, which sends it via LoRa to Gateway #2, which sends it via ESP-NOW to Controller.

Let me know if I'm misunderstanding the situation you're trying to create.

lamasseriadipolverara, 2023-03-30:
> I can't seem to recreate the problem. Tell me if I'm understanding correctly: You are trying to get data from Gateway 0x01 to Gateway 0x02 via LoRa, then from Gateway 0x02 to a Controller via ESP-NOW. When you send serial data to Gateway 0x02, it successfully sends to Controller. However, when you send the data from Gateway 0x01 to Gateway 0x02 via LoRa, Gateway 0x02 does not send the data to Controller via ESP-NOW.
> 
> I copied your configs into a sketch and changed to my own LoRa configs, then I used the 1_UART_Gateway sketch and modified the LoRa neighbor 2 to 0x02, and enabled USE_LORA. With this setup I can send data to the serial monitor of Gateway 0x01, which sends it via LoRa to Gateway 0x02, which sends it via ESP-NOW to Controller.
> 
> Let me know if I'm misunderstanding the situation you're trying to create.

This is my configuration. The only difference (just to be complete) is that the serial of 0x01 comes from 0x00 mqtt gateway to serial to 0x01.
The strange thing is that when i used only lora the message from 0x01 to 0x02 arrives without problem but when i use espnow AND lora in 0x02 the Lora message from 0x01 doesn't arrive

closed 2023-04-25 by timmbogner