timmbogner/Farm-Data-Relay-System #162: MQTT over TLS/SSL

PerRieland, 2023-04-09:
Some cloud MQTT services will only work over TLS/SSL. It would be nice if FDRS had the option to connect encrypted.
I made some small changes to fdrs_gateway_mqtt.h so it could connect via SSL. Maybee not the right way to do it, but it can be used as a starter. It works with EMQX cloud.
Here is what i did in fdrs_gateway_mqtt.h:
1: Add #include <WiFiClientSecure.h>
2: Add a certificate
3: Changed WiFiClient espClient; to WiFiClientSecure espClient;
4: In void begin_mqtt(), set the certificate using espClient.setCACert(ca_cert);

[fdrs_gateway_mqtt.zip](https://github.com/timmbogner/Farm-Data-Relay-System/files/11184963/fdrs_gateway_mqtt.zip)

timmbogner, 2023-04-10:
Very cool, thanks for posting this! There's another user aiming for something like this to use AWS.

Can WiFiClientSecure fully replace regular WiFiClient? If so, I can find a way to make it into an optional element and put it into the official code at some point without much extra fuss.

PerRieland, 2023-04-13:
Not sure if WiFiClientSecure can replace WiFiClient. It compile and upload without any problems. But when trying to use a NON SSL MQTT services, i failed.