timmbogner/Farm-Data-Relay-System #165: Unable to run on ESP32-C3

gco33, 2023-04-20:
I am currently trying to run fdrs gateway on an ESP32-C3.
Compilation is OK, upload is OK but after that, ESP32-C3 fall in bootloop.

Is this a limitation because ESP32-C3 has only one core ?

I had tried ESPNOW examples and was able to run without trouble.

Any advice ?

timmbogner, 2023-04-20:
I tested and also have the same problem on gateways. Node sketches like the ESPNOW_Sensor and ESPNOW_Controller seem to work fine. 

It seems to have been happening for a while, but I'm not sure why at this point. The error happens very early so I'm struggling to pinpoint it. I'll leave this open and post updates when I find more clues.

Thanks for bringing it to my attention!

timmbogner, 2023-04-21:
There's nothing in FDRS's code that utilizes different cores or RTOS explicitly, but it still could be related.

gco33, 2023-04-25:
Hello @bvwelch 

I have tried the following board with 1_UART_Gateway and 2_ESPNOW_Repeater

Information about mcu are from sketch GetChipID from examples

- M5Stack STAMP-C3 (ESP32 Chip model = ESP32-C3 Rev 3, This chip has 1 cores, Chip ID: 10959752)
- Lolin C3-Pico : (ESP32 Chip model = ESP32-C3 Rev 3, This chip has 1 cores, Chip ID: 1813920)
- Adafruitt QT Py ESP32-S2 (ESP32 Chip model = ESP32-S2 Rev 0, This chip has 1 cores, Chip ID: 11233424)

ESP32-C3 seem to boot loop (debug messages in serial debug)

ESP32-S2 does not show any message but become very hot.

Hope it can help to understand what can be wrong.

Thanks.

bvwelch, 2023-05-01:
Greetings,

Some rough notes -- no conclusions yet -- just pasting them here for safe-keeping.

my board is the XIAO_ESP32C3
ESP32 Chip model = ESP32-C3 Rev 4, This chip has 1 cores, Chip ID: 341680   

**************** test of  ESPNOW_Sensor.ino   *******************
FDRS User Node initializing...
 Reading ID 1
 Gateway: 1
Initializing ESP-NOW!
 ESP-NOW Initialized.
Id: 1 - Type: 3 - Data loaded: 55.00
Id: 1 - Type: 1 - Data loaded: 22.07
Sending FDRS Packet!
Nope, not so much.
 Deep sleeping.

(the next line is from my terminal program on my desktop, indicating that the usb serial port disappeared.)
FATAL: read zero bytes from port)

Note: it was very hard to get back to the bootloader -- I had to unplug the USB cable, hold down boot button, then plug in USB cable.

**************** test of  ESPNOW_Sensor.ino   *******************
This test has DEEP_SLEEP disabled in fdrs_node_config.h

FDRS User Node initializing...
 Reading ID 1
 Gateway: 1
Initializing ESP-NOW!
 ESP-NOW Initialized.
Id: 1 - Type: 3 - Data loaded: 40.00
Id: 1 - Type: 1 - Data loaded: 22.07
Sending FDRS Packet!
Nope, not so much.
 Delaying.
Id: 1 - Type: 3 - Data loaded: 79.00
Id: 1 - Type: 1 - Data loaded: 22.07
Sending FDRS Packet!
Nope, not so much.
 Delaying.
Id: 1 - Type: 3 - Data loaded: 52.00
Id: 1 - Type: 1 - Data loaded: 22.07
Sending FDRS Packet!
Nope, not so much.
 Delaying.
Id: 1 - Type: 3 - Data loaded: 83.00
Id: 1 - Type: 1 - Data loaded: 22.07
Sending FDRS Packet!
Nope, not so much.
 Delaying.
Id: 1 - Type: 3 - Data loaded: 64.00
Id: 1 - Type: 1 - Data loaded: 22.07
Sending FDRS Packet!
Nope, not so much.
 Delaying.

(seems OK , I stopped it manually)

Note: I will try other programs, but at this point my theory is something is different about DEEP SLEEP.

timmbogner, 2023-05-01:
>(the next line is from my terminal program on my desktop, indicating that the usb serial port disappeared.)
FATAL: read zero bytes from port)

It sounds like your board is also cutting off the USB interface to save power while it deep-sleeps (not sure, though).

The problem referred to here lies in the gateway code, however. Booting gateways on C3 fails before FDRS even has a chance to start sending UART messages, so troubleshooting will be unusual. Probably the most informative attribute of the bug is that nodes seem to work fine, so that sort of rules out some possible causes.

bvwelch, 2023-05-01:
Yes I see, I will try gateway sketch next

timmbogner, 2023-05-02:
Yep, that's it. 
>By the way, we might consider adding an attach after wakeup from deep sleep:

Not sure what that's doing. Is it something involving native USB?

bvwelch, 2023-05-02:
#if defined(USBCON) && !defined(USE_TINYUSB)
  delay(100);
  USBDevice.attach();
  delay(100);
#endif

Maybe this is just for M0 et al that has built-in USB host controller. Sorry.

bvwelch, 2023-05-02:
I am tired and maybe this is a red herring but I think the issue is Serial1 ...  I took the working  ESPNOW_Sensor.ino sketch and hacked in 3 lines of Serial1 code and now it also crashes at the beginning...

#include "fdrs_node_config.h"
#include <fdrs_node.h>

#define UART_IF Serial1
#define RXD2 14
#define TXD2 15

float data1;
float data2;

void setup() {
  sleep(10);
  UART_IF.begin(115200, SERIAL_8N1, RXD2, TXD2);
  beginFDRS();
}
It is bedtime for Bonzo here.

timmbogner, 2023-05-02:
You may be on to something! I see a couple[ past issues ](https://forum.seeedstudio.com/t/how-to-use-serial1-with-xiao-esp32c3/266306)on Google.

bvwelch, 2023-05-03:
// UART data interface pins (if available)
//#define RXD2 14
//#define TXD2 15

#define RXD2 9 // 18 // 14
#define TXD2 10 // 19 //15

This works for me!

Apparently either 14 or 15 clashes with something and causes problems.

Type [C-a] [C-h] to see available commands
Terminal ready
Address:1
Initializing ESP-NOW!
[ 10148][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 0 - WIFI_READY
[ 10180][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[ 10181][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 2 - STA_START
[ 10183][V][WiFiGeneric.cpp:343] _arduino_event_cb(): STA Stopped
[ 10188][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 3 - STA_STOP
[ 10195][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[ 10201][D][WiFiGeneric.cpp:1035] _eventCallback(): Arduino Event: 2 - STA_START
 ESP-NOW Initialized.

timmbogner, 2023-05-03:
Well gosh that's a really easy fix! I forget which board the default pins (14 & 15) were chosen for, but we'll have to find a pair that works more universally.

Thanks a ton for doing the dirty work... it's a funny feeling when an issue turns out to be just a pin conflict. On one hand it's nice that it wasn't a system problem, but frustrating that it took so long to troubleshoot such a small error. In hindsight, it was so obvious: "What's failing before UART??" "Duhhh... the other UART!" 😅

gco33, 2023-05-03:
Hello,

I was able to run gateway on my esp-c3 (lolin c3 mini) but not on the esp32-s2.
I have put same pins for RX2 and TX2 as @bvwelch mentionned in his post.

Thanks for investigation.

To solve issue, may be some conditional defines can select correct pins depending on mpu used.

timmbogner, 2023-05-03:
Confirmed: As per [the ESP32-C3 docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/gpio.html):
>SPI0/1: GPIO12-17 are usually used for SPI flash and PSRAM and are not recommended for other uses.

@gco33 Here's [the same doc for the S2](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/gpio.html). Perhaps try GPIO 33-38?

bvwelch, 2023-05-03:
I dont know if it would help, but when I hovered over the calls to the serial api, it directed me to the actual esp32 library source code, and it seemed like if you leave off the pins, it will default to (hopefully) reasonable values.

gco33, 2023-05-11:
I finally had time to test modification of RX2/TX2 pins on my qt-py esp32-s2 and was able to run gateway.

Thanks

shaffenmeister, 2023-08-13:
I use the following preprocessor directives in fdrs_gateway_config.h to run FDRS gateways with ESP32, ESP32-S2, ESP32-S3, ESP32-C3:
```
#if !defined RXD2 or !defined TXD2
    #warning Defining RXD2 and TXD2 using MCU defaults.
    #if CONFIG_IDF_TARGET_ESP32
        #define RXD2 14
        #define TXD2 15
    #elif CONFIG_IDF_TARGET_ESP32S2 or CONFIG_IDF_TARGET_ESP32S3
        #define RXD2 18
        #define TXD2 17
    #elif CONFIG_IDF_TARGET_ESP32C3
        #define RXD2 2
        #define TXD2 3
    #else
        #error MCU not supported.
    #endif
#endif

```
These are safe defaults for the respective MCU and may not fit to a specific board.

For board specific adaptations, one could adapt above defines or deploy build options, e.g. in platformio.ini:
```
[env:seeed_xiao_esp32s3]
platform = espressif32@^6.3.2
board = seeed_xiao_esp32s3
board_build.mcu = esp32s3
board_build.f_cpu = 240000000L
build_flags =
    '-D RXD2=18'
    '-D TXD2=17'
```

timmbogner, 2023-09-08:
Thanks @shaffenmeister and @bvwelch for the assistance! I have implemented this now. Let me know if there are any snags.

closed 2023-09-08 by timmbogner