timmbogner/Farm-Data-Relay-System #168: Upgrade to RadioLib 6.0.0

Stips5, 2023-05-01:
Hi

I found issue when using RadioLib 6.0.0 which is latest lib version

`//Farm-Data-Relay-System/src/fdrs_gateway_lora.h: In function 'void begin_lora()':
/Arduino/libraries/Farm-Data-Relay-System/src/fdrs_gateway_lora.h:293:30: error: no matching function for call to 'SX1278::setDio0Action(void (&)())'
   radio.setDio0Action(setFlag);
                              ^
In file included from /Arduino/libraries/RadioLib/src/modules/RFM9x/RFM95.h:9:0,
                 from /Arduino/libraries/RadioLib/src/RadioLib.h:77,
                 from /Arduino/libraries/Farm-Data-Relay-System/src/fdrs_gateway_lora.h:1,
                 from /Arduino/libraries/Farm-Data-Relay-System/src/fdrs_gateway.h:76,
                 from /.git/MedjuglaviceAtPalm/RemoteGateway/RemoteGateway.ino:9:
Arduino/libraries/RadioLib/src/modules/RFM9x/../SX127x/SX127x.h:700:10: note: candidate: void SX127x::setDio0Action(void (*)(), uint32_t)
     void setDio0Action(void (*func)(void), uint32_t dir);
          ^
Arduino/libraries/RadioLib/src/modules/RFM9x/../SX127x/SX127x.h:700:10: note:   candidate expects 2 arguments, 1 provided

exit status 1`

when using 5.7.0 everything works just fine. So project should be updated or restriction to use 5.7.0 specified in readme.

timmbogner, 2023-05-01:
Thanks! I just saw the RadioLib release notes about this yesterday, and fixing it is on my big board of things to do. It's kind of urgent, so I'm going to have to get into it this tomorrow along with a whole backlog of code updates.

timmbogner, 2023-05-03:
I added the extra parameter in a commit last night, so I think everything should work okay now. I'm going to leave this open until I'm a bit more confident about the other RL changes.

It also still works with RL v5.7, which is a relief to me.

PerRieland, 2023-05-03:
Issue still exist when using 
#define RADIOLIB_MODULE SX1262
#define USE_SX126X

I'm not sure how to fix that. 
Compiler output:
c:\users\xxx\documents\arduino\libraries\radiolib\src\modules\sx126x\SX126x.h:568:10: note: candidate: 'virtual void SX126x::setDio1Action(void (*)())'
     void setDio1Action(void (*func)(void));
          ^~~~~~~~~~~~~
c:\users\xxx\documents\arduino\libraries\radiolib\src\modules\sx126x\SX126x.h:568:10: note:   candidate expects 1 argument, 2 provided
exit status 1
Compilation error: exit status 1

timmbogner, 2023-05-03:
Very good point. It looks like the new thing is only applicable to SX127x. I've now reversed the change for that chip.

aviateur17, 2023-05-04:
> @aviateur17 Could you maybe help explain what this second point means? If I'm understanding right, it's not an issue unless we're working with a pin value that was returned by RadioLib or assigning a pin based using a variable (which we never do).

Looking at the commit (https://github.com/jgromes/RadioLib/commit/9a68a3c9012e1ffe5dc94cb4d116cbe71ad3d3ea)  all they really do is change uint8_t to uint32_t so the only thing would be to check our types to make sure they are uint32_t otherwise there may be a compiler warning.  I don't see a need to adjust any pin names or values just because of this change.

PerRieland, 2023-05-04:
> Very good point. It looks like the new thing is only applicable to SX127x. I've now reversed the change for that chip.

And it works ;)

closed 2023-05-04 by timmbogner