timmbogner/Farm-Data-Relay-System #179: sleepFDRS does not go into 'Deep sleeping' at values above 2147

FMJK190, 2023-07-26:
I experience sleepFDRS does not go into 'Deep sleeping' at values above 2147.

DEEP_SLEEP is enabled: #define DEEP_SLEEP

aviateur17, 2023-07-26:
@FMJK190, which value are you setting above 2147?  Can you give a copy of that line of code?
Thanks!

FMJK190, 2023-07-27:
@aviateur17 I have only tried with values up to 3600.
You can try with 2148, here the command does not go into deep sleep.
Here is a line of code:
  sleepFDRS(2148);  

my environment is: PlatformIO Core 6.1.9.

Here's a small example that doesn't go to sleep. Changes sleepFDRS(2148); to sleepFDRS(2147); then it go to sleep.
#include "fdrs_node_config.h"
#include <fdrs_node.h>

void setup() {
  beginFDRS();
}

void loop() {
  delay(50); //let the sensor warm up
  int sensorValue = analogRead(A0);
  Serial.println(sensorValue);
  const int waterValue  = 1677;
  const int airValue  = 3358;
  int soilMoisturePercent  = map(sensorValue,airValue ,waterValue ,0,100);
  
  loadFDRS(soilMoisturePercent, SOIL_T);
  
  if(sendFDRS()){
    DBG("Big Success!");
  } else {
    DBG("Nope, not so much.");
  }
  sleepFDRS(2148);  //Sleep time in seconds  
}

timmbogner, 2023-07-28:
Hey guys, sorry I've been so absent! 

It sounds like [this issue. ](https://github.com/espressif/arduino-esp32/issues/3286) 

Definitely an oversight on my part, it needs to be working with a 64bit number there. 

It can probably be fixed by adding `ULL` to [this line.](https://github.com/timmbogner/Farm-Data-Relay-System/blob/5456c0378bb7281311821236d90a9c9d6f6e31ef/src/fdrs_node.h#L257) 

I've been taking a bit of a hiatus to focus on outdoor stuff. If you would like to fix it and submit a pull request, you're welcome to! Otherwise I will get to it when I can, and leave this issue open until I do. 

Thanks for reporting this!

egzumer, 2023-07-29:
`sleep_time` should be unsigned. Negative values doesn't make sense, and none of the subsequent sleep/delay functions take signed types.

timmbogner, 2023-07-30:
Good call, @egzumer ! There are probably several int types within the project that have existed since before I knew better. I will add "Fix all ints" to my list of tasks to do. 

Thanks!

timmbogner, 2023-10-01:
I think I've fixed this issue plus the int issue. Please let me know if the problem persists.

Thanks to everyone in this thread!

closed 2023-10-17 by timmbogner