timmbogner/Farm-Data-Relay-System #186: Sketch BMP280_fdrs.ino compile fails with #USE_LORA defined

NorthBaandee, 2023-10-31:
Board: TTGP LoRa32 OLED

#####################################################################
error messages:

In file included from /home/peter/Arduino/libraries/Farm-Data-Relay-System/src/fdrs_node.h:68,
                 from /home/peter/Arduino/libraries/Farm-Data-Relay-System/examples/Sensor_Examples/BMP280_fdrs/BMP280_fdrs.ino:10:
/home/peter/Arduino/libraries/Farm-Data-Relay-System/src/fdrs_node_lora.h:82:65: error: 'LORA_BUSY' was not declared in this scope
 RADIOLIB_MODULE radio = new Module(LORA_SS, LORA_DIO, LORA_RST, LORA_BUSY);
                                                                 ^~~~~~~~~
/home/peter/Arduino/libraries/Farm-Data-Relay-System/src/fdrs_node_lora.h:82:65: note: suggested alternative: 'LORA_CS'
 RADIOLIB_MODULE radio = new Module(LORA_SS, LORA_DIO, LORA_RST, LORA_BUSY);
                                                                 ^~~~~~~~~
                                                                 LORA_CS
/home/peter/Arduino/libraries/Farm-Data-Relay-System/src/fdrs_node_lora.h:85:6: error: redefinition of 'bool pingFlag'
 bool pingFlag = false;
      ^~~~~~~~
In file included from /home/peter/Arduino/libraries/Farm-Data-Relay-System/src/fdrs_node.h:65,
                 from /home/peter/Arduino/libraries/Farm-Data-Relay-System/examples/Sensor_Examples/BMP280_fdrs/BMP280_fdrs.ino:10:
/home/peter/Arduino/libraries/Farm-Data-Relay-System/src/fdrs_node_espnow.h:13:6: note: 'bool pingFlag' previously defined here
 bool pingFlag = false;
      ^~~~~~~~
In file included from /home/peter/Arduino/libraries/Farm-Data-Relay-System/examples/Sensor_Examples/BMP280_fdrs/BMP280_fdrs.ino:10:
/home/peter/Arduino/libraries/Farm-Data-Relay-System/src/fdrs_node.h: In function 'uint32_t pingFDRS(uint32_t)':
/home/peter/Arduino/libraries/Farm-Data-Relay-System/src/fdrs_node.h:333:12: error: redeclaration of 'uint32_t pingResponseMs'
   uint32_t pingResponseMs = pingFDRSLoRa(&gtwyAddress, timeout);
            ^~~~~~~~~~~~~~
/home/peter/Arduino/libraries/Farm-Data-Relay-System/src/fdrs_node.h:329:12: note: 'uint32_t pingResponseMs' previously declared here
   uint32_t pingResponseMs = pingFDRSEspNow(gatewayAddress, timeout);
            ^~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1

#######################################################
fdrs_node_config.h contents:

//  FARM DATA RELAY SYSTEM
//
//  Sensor Configuration

#define READING_ID    1   //Unique ID for this sensor
#define GTWY_MAC      0x02 //Address of the nearest gateway
#define HASL          280
#define P_CORRECTION  HASL * 0.1222

// OLED -- Displays console debugging messages on an SSD1306 I²C OLED
#define USE_OLED    
#define OLED_HEADER "FDRS"
#define OLED_PAGE_SECS 30
#define OLED_SDA 4
#define OLED_SCL 15
#define OLED_RST 16

#define USE_ESPNOW
#define USE_LORA
#define DEEP_SLEEP
//#define POWER_CTRL    14
#define FDRS_DEBUG
// LoRa Configuration
#define RADIOLIB_MODULE SX1276 //Tested on SX1276
#define LORA_SS 18
#define LORA_RST 14
#define LORA_DIO 26

#define LORA_TXPWR 17    // LoRa TX power in dBm (: +2dBm - +17dBm (for SX1276-7) +20dBm (for SX1278))
#define LORA_ACK        // Request LoRa acknowledgment.

######################################################################
BMP280.ino contents:

//  FARM DATA RELAY SYSTEM
//
//  BMP280 SENSOR MODULE
//
//  Developed by Timm Bogner (timmbogner@gmail.com) in Urbana, Illinois, USA.
//  Connect sensor SDA and SCL pins to those of the ESP.

#include "fdrs_node_config.h"
#include <Adafruit_BMP280.h>
#include <fdrs_node.h>

Adafruit_BMP280 bmp;

void setup() {
  //Serial.begin(115200);
  beginFDRS();
  while (!bmp.begin(0x76)) {
    //Serial.println("BMP not initializing!");
    delay(10);
  }
}

void loop() {
  loadFDRS(bmp.readTemperature(), TEMP_T);
  loadFDRS((bmp.readPressure() / 100.0F) + P_CORRECTION, PRESSURE_T);
  sendFDRS();
  sleepFDRS(60);  //Sleep time in seconds
}

NorthBaandee, 2023-10-31:
It was necessary to add #define LORA_BUSY 33 to fdrs_node_config.h, which then produced another error caused by having both USE_LORA and USE_ESPNOW defined. I commented out #define USE_ESPNOW which then produced yet another error which was resolved by changing a line in fdrs_node.h
The offending line was
if (is_controller){
  handleIncoming();
which I changed to 
if (isControl){
  handleIncoming();

Finally, by adding these two lines to fdrs_node_config.h:
#define HASL          280
#define P_CORRECTION  HASL * 0.1222
and changing this line in BMP280_fdrs.ino:
 loadFDRS(bmp.readPressure() / 100.0F, PRESSURE_T);
to this:
 loadFDRS((bmp.readPressure() / 100.0F) + P_CORRECTION, PRESSURE_T);
I am able to get the correct MSLP for the sensor

cheers
PeterB

timmbogner, 2023-11-02:
Thanks for the report, Peter! The first issue was on my end: The SX1278 chip doesn't need the LORA_BUSY pin to be defined, so on gateways, I have code to define it if the user does not. I had forgotten to add that functionality to nodes, and just fixed it.

Next, a node may use either ESP-NOW or LoRa, but not both. I'm not sure if I documented that somewhere or not, but I'll make a note of it in the node readme. 

I think your final problem will be fixed by getting the latest copy of the repo, as I recently fixed the `is_controller` issue.

`isControl` is an [Arduino function](https://www.arduino.cc/reference/en/language/functions/characters/iscontrol/) and is unrelated to FDRS.

Let me know if this irons things out!

NorthBaandee, 2023-11-02:
Your code is very nicely written and therefore easy to troubleshoot. I understand the architecture much better now. I'm looking at things like NPK probes,  ultrasonic tank level sensing, reed switch gate monitors and stepper motor tuning knobs. small servos also have some farm applications.

timmbogner, 2023-11-03:
I appreciate it, but must pass the credit to various contributors for their guidance in organizing the code and repo. This thing was a mess back when it was just me 😅

If you're working with a sensor that is not covered in the FDRS examples, please feel encouraged to contribute an example sketch! 
I'm a bit surprised that there isn't an ultrasonic sensor sketch. I know I've used one with FDRS before, so I'll have to look for it. 
I'm going to be using a servo to tip a rain gauge I built from this[ cool project](https://hackaday.io/project/191652-long-range-weather-station-65) but I haven't had a chance to do the example code.

NorthBaandee, 2023-11-03:
Farmers love rain gauges ;)