timmbogner/Farm-Data-Relay-System #187: Datatype TEXT

hakha4, 2023-11-23:
Hi!
Stumbled over FRDS and It looks very nice,Got a basic setup to work without issues. I'm working on a ESPnow node with mcp23017 to manage multiple relays and it seems to work fine but I miss a datatype TEXT though, I want to send relaystatus as a binary string like '001100110010011' from node for further manipulation. Is it hard to add this datatype?
Best Regards Håkan

timmbogner, 2024-01-27:
This issue was discussed further in #188. I've decided to switch to CBOR to encode the data portion of the DataReading object. My current method of moving all data as floats is not ideal, and I've never liked it. I'm leaving this issue open until I make those changes.

If anyone has questions or suggestions regarding this topic, this is the place for it!

bvwelch, 2024-05-13:
I've used cbor on a project or two. It wasn't so bad. I used a library named tinycbor, and I also wrote a small library straight from the rfc.  lemme know if i can help.