timmbogner/Farm-Data-Relay-System #209: Fixed a problem with possibly new espressif api

kheinz57, 2024-06-02:
I got some compile messages which I have fixed with this patch. 
As I have no real board yet I can not proof the fix but the compiler at least succeeded.
The function signature for the callback has changed in its first parameter from mac_addr to esp_now_recv_info

diff --git a/src/fdrs_gateway_espnow.h b/src/fdrs_gateway_espnow.h
index cf1513c..3d08056 100644
--- a/src/fdrs_gateway_espnow.h
+++ b/src/fdrs_gateway_espnow.h
@@ -41,10 +41,11 @@ void OnDataSent(const uint8_t *mac_addr, esp_now_send_status_t status)
 {
 esp_now_sent_flag = true;
 }
-void OnDataRecv(const uint8_t *mac, const uint8_t *incomingData, int len)
+//                const uint8_t *mac_addr, const uint8_t *data, int data_len)
+void OnDataRecv(const esp_now_recv_info *mac, const uint8_t *incomingData, int len)
 {
 #endif
-  memcpy(&incMAC, mac, sizeof(incMAC));
+  memcpy(&incMAC, mac->src_addr, sizeof(incMAC));
   if (len < sizeof(DataReading))
   {
     DBG1("Incoming ESP-NOW System Packet from 0x" + String(incMAC[5], HEX));
diff --git a/src/fdrs_node_espnow.h b/src/fdrs_node_espnow.h
index 64b6a3f..d61e1eb 100644
--- a/src/fdrs_node_espnow.h
+++ b/src/fdrs_node_espnow.h
@@ -80,10 +80,11 @@ void OnDataSent(const uint8_t *mac_addr, esp_now_send_status_t status)
         esp_now_ack_flag = CRC_BAD;
     }
 }
-void OnDataRecv(const uint8_t *mac, const uint8_t *incomingData, int len)
+
+void OnDataRecv(const esp_now_recv_info *mac, const uint8_t *incomingData, int len)
 {
 #endif
-    memcpy(&incMAC, mac, sizeof(incMAC));
+    memcpy(&incMAC, mac->src_addr, sizeof(incMAC));
     if (len == sizeof(SystemPacket))
     {
         SystemPacket command;

timmbogner, 2024-06-02:
I'd heard murmurs about a breaking change to ESP-NOW, but when I updated my ESP32 boards in Arduino yesterday it still worked. Today I see it did not update to the newest 3.0 version. I'm updating now and I can see the changes [from the documentation](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_now.html#_CPPv417esp_now_recv_info).

Thanks a lot for submitting this! I'll get it patched up soon and tack you on as a co-contributor!

kheinz57, 2024-06-02:
Yes I have installed version 3.0.0 today.
BTW: I am trying to compile it first and since it seems to compile I will order 2 lora boards (https://www.lilygo.cc/products/lora3?variant=42272562282677)
I would have bought esp32-s3 boards but not sure if those will work

Thanks
Heinz

timmbogner, 2024-06-02:
I also just purchased that same LoRa32 V2.1_1.6 yesterday. There shouldn't be any issue!
I bought their T3-S3 LoRa boards last year and I'm afraid they're a bit of a pain to develop on so far. I think it's largely due to some issues between Arduino's COM port handler and the ESP32S3's native USB. Perhaps the 3.0 update will address some of them. LilyGo did send me samples of the T3-S3 with 2.4Ghz LoRa, but I still haven't had a chance to work with them.

timmbogner, 2024-06-09:
Thanks again, @kheinz57!

closed 2024-06-09 by timmbogner

aviateur17, 2024-06-14:
Well that was fun.  For those of us using VSCode and Platform IO it seems that the PIO team has not adopted the new 3.0 version in the stable branch.  You have to go to the development branch of PIO.   I had to make some changes to platformio.ini to get the new development branch to work:

```
platform = https://github.com/platformio/platform-espressif32.git
platform_packages=
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.1
  framework-arduinoespressif32-libs @ https://github.com/espressif/arduino-esp32/releases/download/3.0.1/esp32-arduino-libs-3.0.1.zip
```
Now when I run the compile, in the output I see:

```
PACKAGES: 
 - framework-arduinoespressif32 @ 3.0.1+sha.cc50d90 
 - framework-arduinoespressif32-libs @ 5.1.0+sha.442a798083 
```

Using the stable version the same output was:
```
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20016.0 (2.0.16) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
```

For more information see:
https://github.com/platformio/platform-espressif32/issues/1225

aviateur17, 2024-06-14:
Seems to be a political or monetary issue between Espressif and PlatformIO Teams.

https://github.com/platformio/platform-espressif32/issues/1225#issuecomment-2143898558

For PIO users I'm going to paste my compile errors so that this can be, hopefully, found a bit quicker.

```
In file included from src/fdrs_gateway.h:83,
                 from src/host/main_esp32s3.h:2,
                 from src/host/host_8eb5d0.h:117,
                 from src/main.cpp:41:
src/fdrs_gateway_espnow.h:45:23: error: 'esp_now_recv_info' does not name a type; did you mean 'esp_now_peer_info'?
 void OnDataRecv(const esp_now_recv_info *pkt_info, const uint8_t *incomingData, int len)
                       ^~~~~~~~~~~~~~~~~
                       esp_now_peer_info
src/fdrs_gateway_espnow.h: In function 'void OnDataRecv(const int*, const uint8_t*, int)':
src/fdrs_gateway_espnow.h:47:29: error: request for member 'src_addr' in '* pkt_info', which is of non-class type 'const int'
   memcpy(&incMAC, pkt_info->src_addr, sizeof(incMAC));
                             ^~~~~~~~
src/fdrs_gateway_espnow.h: In function 'void begin_espnow()':
src/fdrs_gateway_espnow.h:106:28: error: invalid conversion from 'void (*)(const int*, const uint8_t*, int)' {aka 'void (*)(const int*, const unsigned char*, int)'} to 'esp_now_recv_cb_t' {aka 'void (*)(const unsigned char*, const unsigned char*, int)'} [-fpermissive]
   esp_now_register_recv_cb(OnDataRecv);
                            ^~~~~~~~~~
In file included from src/fdrs_gateway_espnow.h:5,
                 from src/fdrs_gateway.h:83,
                 from src/host/main_esp32s3.h:2,
                 from src/host/host_8eb5d0.h:117,
                 from src/main.cpp:41:
/home/jlehman/.platformio/packages/framework-arduinoespressif32@3.20016.0/tools/sdk/esp32s3/include/esp_wifi/include/esp_now.h:136:54: note:   initializing argument 1 of 'esp_err_t esp_now_register_recv_cb(esp_now_recv_cb_t)'
 esp_err_t esp_now_register_recv_cb(esp_now_recv_cb_t cb);
                                    ~~~~~~~~~~~~~~~~~~^~
```

timmbogner, 2024-06-15:
@aviateur17 If this continues, can we use directives to compile ESP-NOW differently when under ESP32 core v2.x? Or, alternatively to compile differently for PIO if needed? That's my best solution at the moment. Hopefully it's just Espressif dragging their feet.  

Definitely sounds like a pain. Did you get it working?

aviateur17, 2024-06-15:
Yeah, got it working so I don't think we need to take any action now.  Hopefully Espressif and Platform IO get things figured out or many people will be disappointed.   I just wanted to make a note for the others that are using PlatformIO on the steps they need to get it working.    The issue is due to the change in Espressif API in their more up to date versions and PlatformIO not supporting those versions in stable branch yet.  Hope they get it figured out.

timmbogner, 2024-06-15:
Yep, just read up on it and it sounds like a business disagreement. I didn't actually realize that Espressif supports their PIO core, I guess I assumed the PIO community modified the Arduino core.