timmbogner/Farm-Data-Relay-System #211: Compilation error on "enableInterrupt": redeclared as different kind of entity

PerRieland, 2024-06-04:
Not sure how what cause this and how to fix the compilation error. Any help is appreciated

In file included from c:\Users\Per\Documents\Arduino\libraries\Farm-Data-Relay-System\src/fdrs_node.h:71,
                 from G:\My Drive\dev\V4\Irrigation_VG_6000\Irrigation_VG_6000.ino:9:
c:\Users\Per\Documents\Arduino\libraries\Farm-Data-Relay-System\src/fdrs_node_lora.h:87:6: error: 'bool enableInterrupt' redeclared as different kind of entity
   87 | bool enableInterrupt = true; // disable interrupt when it's not needed
      |      ^~~~~~~~~~~~~~~
In file included from C:\Users\Per\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\cores\esp32/esp32-hal.h:75,
                 from C:\Users\Per\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\cores\esp32/Arduino.h:36,
                 from D:\Temp\arduino\sketches\CF1240978213A8A79D16F9BC678FB6AA\sketch\Irrigation_VG_6000.ino.cpp:1:
C:\Users\Per\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\cores\esp32/esp32-hal-gpio.h:80:6: note: previous declaration 'void enableInterrupt(uint8_t)'
   80 | void enableInterrupt(uint8_t pin);
      |      ^~~~~~~~~~~~~~~
c:\Users\Per\Documents\Arduino\libraries\Farm-Data-Relay-System\src/fdrs_node_lora.h: In function 'crcResult handleLoRa()':
c:\Users\Per\Documents\Arduino\libraries\Farm-Data-Relay-System\src/fdrs_node_lora.h:122:25: error: assignment of function 'void enableInterrupt(uint8_t)'
  122 |         enableInterrupt = false;
      |         ~~~~~~~~~~~~~~~~^~~~~~~
c:\Users\Per\Documents\Arduino\libraries\Farm-Data-Relay-System\src/fdrs_node_lora.h:128:29: error: assignment of function 'void enableInterrupt(uint8_t)'
  128 |             enableInterrupt = true;
      |             ~~~~~~~~~~~~~~~~^~~~~~
c:\Users\Per\Documents\Arduino\libraries\Farm-Data-Relay-System\src/fdrs_node_lora.h:138:29: error: assignment of function 'void enableInterrupt(uint8_t)'
  138 |             enableInterrupt = true;
      |             ~~~~~~~~~~~~~~~~^~~~~~

exit status 1

Compilation error: exit status 1

lamasseriadipolverara, 2024-06-04:
similar here
In file included from C:\Users\utente\Documents\Arduino\libraries\Farm-Data-Relay-System\src/fdrs_gateway.h:86,
                 from C:\Users\utente\Documents\Arduino\lora serra\2_LoRa_Repeater\2_LoRa_Repeater.ino:9:
C:\Users\utente\Documents\Arduino\libraries\Farm-Data-Relay-System\src/fdrs_lora.h:138:15: error: 'volatile bool enableInterrupt' redeclared as different kind of entity
  138 | volatile bool enableInterrupt = true; // disable interrupt when it's not needed
      |               ^~~~~~~~~~~~~~~
In file included from C:\Users\utente\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\cores\esp32/esp32-hal.h:75,
                 from C:\Users\utente\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\cores\esp32/Arduino.h:36,
                 from C:\Users\utente\AppData\Local\Temp\arduino\sketches\54BA3C4527A270946030C9602DB79DDD\sketch\2_LoRa_Repeater.ino.cpp:1:
C:\Users\utente\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\cores\esp32/esp32-hal-gpio.h:80:6: note: previous declaration 'void enableInterrupt(uint8_t)'
   80 | void enableInterrupt(uint8_t pin);
      |      ^~~~~~~~~~~~~~~
C:\Users\utente\Documents\Arduino\libraries\Farm-Data-Relay-System\src/fdrs_lora.h: In function 'crcResult LoRaTxRxOperation()':
C:\Users\utente\Documents\Arduino\libraries\Farm-Data-Relay-System\src/fdrs_lora.h:641:25: error: assignment of function 'void enableInterrupt(uint8_t)'
  641 |         enableInterrupt = false;
      |         ~~~~~~~~~~~~~~~~^~~~~~~
C:\Users\utente\Documents\Arduino\libraries\Farm-Data-Relay-System\src/fdrs_lora.h:666:25: error: assignment of function 'void enableInterrupt(uint8_t)'
  666 |         enableInterrupt = true;
      |         ~~~~~~~~~~~~~~~~^~~~~~

exit status 1

Compilation error: exit status 1

lamasseriadipolverara, 2024-06-04:
As i see some error with esp32 board version 3 (that i just installed for upgrading everything to the new version before installing in real world), I downgrade to version board esp32 2.0.17, others error comes

In file included from C:\Users\utente\Documents\Arduino\libraries\Farm-Data-Relay-System\src/fdrs_gateway.h:83,
                 from C:\Users\utente\Documents\Arduino\lora serra\2_LoRa_Repeater\2_LoRa_Repeater.ino:9:
C:\Users\utente\Documents\Arduino\libraries\Farm-Data-Relay-System\src/fdrs_gateway_espnow.h:45:23: error: 'esp_now_recv_info' does not name a type; did you mean 'esp_now_peer_info'?
 void OnDataRecv(const esp_now_recv_info *pkt_info, const uint8_t *incomingData, int len)
                       ^~~~~~~~~~~~~~~~~
                       esp_now_peer_info
C:\Users\utente\Documents\Arduino\libraries\Farm-Data-Relay-System\src/fdrs_gateway_espnow.h: In function 'void OnDataRecv(const int*, const uint8_t*, int)':
C:\Users\utente\Documents\Arduino\libraries\Farm-Data-Relay-System\src/fdrs_gateway_espnow.h:47:29: error: request for member 'src_addr' in '* pkt_info', which is of non-class type 'const int'
   memcpy(&incMAC, pkt_info->src_addr, sizeof(incMAC));
                             ^~~~~~~~
C:\Users\utente\Documents\Arduino\libraries\Farm-Data-Relay-System\src/fdrs_gateway_espnow.h: In function 'void begin_espnow()':
C:\Users\utente\Documents\Arduino\libraries\Farm-Data-Relay-System\src/fdrs_gateway_espnow.h:106:28: error: invalid conversion from 'void (*)(const int*, const uint8_t*, int)' {aka 'void (*)(const int*, const unsigned char*, int)'} to 'esp_now_recv_cb_t' {aka 'void (*)(const unsigned char*, const unsigned char*, int)'} [-fpermissive]
   esp_now_register_recv_cb(OnDataRecv);
                            ^~~~~~~~~~
In file included from C:\Users\utente\Documents\Arduino\libraries\Farm-Data-Relay-System\src/fdrs_gateway_espnow.h:5,
                 from C:\Users\utente\Documents\Arduino\libraries\Farm-Data-Relay-System\src/fdrs_gateway.h:83,
                 from C:\Users\utente\Documents\Arduino\lora serra\2_LoRa_Repeater\2_LoRa_Repeater.ino:9:
C:\Users\utente\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.17/tools/sdk/esp32/include/esp_wifi/include/esp_now.h:136:54: note:   initializing argument 1 of 'esp_err_t esp_now_register_recv_cb(esp_now_recv_cb_t)'
 esp_err_t esp_now_register_recv_cb(esp_now_recv_cb_t cb);
                                    ~~~~~~~~~~~~~~~~~~^~

exit status 1

Compilation error: exit status 1

timmbogner, 2024-06-04:
At first glance this appears to be another API change from Espressif. I'll be able to investigate further this evening.

timmbogner, 2024-06-04:
I threw together a quick fix, but as a rule I don't publish commits to the main branch before I've had coffee 😄  [Please try out this new branch and let me know if it works!](https://github.com/timmbogner/Farm-Data-Relay-System/tree/enableInterrupt_fix)

lamasseriadipolverara, 2024-06-04:
jajaja
seems to work. Tomorrow some test.
Thank you timm

PerRieland, 2024-06-07:
> I threw together a quick fix, but as a rule I don't publish commits to the main branch before I've had coffee 😄 [Please try out this new branch and let me know if it works!](https://github.com/timmbogner/Farm-Data-Relay-System/tree/enableInterrupt_fix)

Yeaa, this fix is working for me ;)   - Thank's a lot

PerRieland, 2024-06-08:
> I threw together a quick fix, but as a rule I don't publish commits to the main branch before I've had coffee 😄 [Please try out this new branch and let me know if it works!](https://github.com/timmbogner/Farm-Data-Relay-System/tree/enableInterrupt_fix)

What the hell have you done with this update, the system works insanely fast now. Before I experienced approx. 2 sec. delay when I turned something on or off. Now it works instantly. The gateway seems very aggressive ;)
Big big thank you

timmbogner, 2024-06-09:
:D I'm guessing it's because LoRa sends data instantly now instead of delaying. You can thank @aviateur17 for that!

closed 2024-06-09 by timmbogner