timmbogner/Farm-Data-Relay-System #213: Option to define values outside fdrs_globals.h

Stips5, 2024-06-11:
Hi
I would like to define wifi, mqtt... etc private values outside fdrs_globals.h, I found it unpractical this way. I use to have my fdrs_config file where i have mqtt stuff, also I have multiple network config files where my wifi pass and ssid are saved, which i could easily switch in define. Now if am not wrong i'll have to define all of that in globals, which are inside fdr_gateways and those files are fdrs framework code, which I wouldnt like to change in my fork because I dont want to push that data to my branch, but I want to have them stored somewhere.

So I am wondering  what do you think about giving option to pick config, so multiple projects can have same fdrs core, and different wifi networks, mqtt topics etc.

timmbogner, 2024-06-11:
Hey @Stips5 ,
Happy to say you already have this option! Within the 'fdrs_gateway_config.h' you will see a lot of defines commented out:
``` cpp
//#define WIFI_SSID   "Your SSID"  
//#define WIFI_PASS   "Your Password"

...

//#define MQTT_ADDR   "192.168.0.8"
//#define MQTT_PORT   1883 // Default MQTT port is 1883
//#define MQTT_AUTH   //Enable MQTT authentication 
//#define MQTT_USER   "Your MQTT Username"
//#define MQTT_PASS   "Your MQTT Password"
```
By uncommenting any of these lines, you can tell FDRS to ignore the globally defined variable and to use the one you've defined there instead. For example, :

``` cpp
#define WIFI_SSID   "MyRouter"  
#define WIFI_PASS   "qwerty123"

...

#define MQTT_ADDR   "192.168.2.4"
//#define MQTT_PORT   1883 // Default MQTT port is 1883
//#define MQTT_AUTH   //Enable MQTT authentication 
//#define MQTT_USER   "Your MQTT Username"
//#define MQTT_PASS   "Your MQTT Password"
```
Hope this helps!

Stips5, 2024-06-12:
But I faced following issue.

```
In file included from /home/Arduino/libraries/Farm-Data-Relay-System/src/fdrs_gateway.h:7,
                 from /home/.git/proj/LoraHomeGateway_WithMQTT/LoraHomeGateway_WithMQTT.ino:12:
/home/Arduino/libraries/Farm-Data-Relay-System/src/fdrs_globals.h:25: warning: "TOPIC_DATA" redefined
   25 | #define TOPIC_DATA    "fdrs/data"
      | 
In file included from /home/.git/proj/LoraHomeGateway_WithMQTT/LoraHomeGateway_WithMQTT.ino:11:
/home/.git/proj/LoraHomeGateway_WithMQTT/fdrs_gateway_config.h:60: note: this is the location of the previous definition
   60 | #define TOPIC_DATA "proj/data"
      | 
In file included from /home/Arduino/libraries/Farm-Data-Relay-System/src/fdrs_gateway.h:7,
                 from /home/.git/proj/LoraHomeGateway_WithMQTT/LoraHomeGateway_WithMQTT.ino:12:
/home/Arduino/libraries/Farm-Data-Relay-System/src/fdrs_globals.h:26: warning: "TOPIC_STATUS" redefined
   26 | #define TOPIC_STATUS  "fdrs/status"
      | 
In file included from /home/.git/proj/LoraHomeGateway_WithMQTT/LoraHomeGateway_WithMQTT.ino:11:
/home/.git/proj/LoraHomeGateway_WithMQTT/fdrs_gateway_config.h:61: note: this is the location of the previous definition
   61 | #define TOPIC_STATUS "proj/status"
      | 
In file included from /home/Arduino/libraries/Farm-Data-Relay-System/src/fdrs_gateway.h:7,
                 from /home/.git/proj/LoraHomeGateway_WithMQTT/LoraHomeGateway_WithMQTT.ino:12:
/home/Arduino/libraries/Farm-Data-Relay-System/src/fdrs_globals.h:27: warning: "TOPIC_COMMAND" redefined
   27 | #define TOPIC_COMMAND "fdrs/command"
      | 
In file included from /home/.git/proj/LoraHomeGateway_WithMQTT/LoraHomeGateway_WithMQTT.ino:11:
/home/.git/proj/LoraHomeGateway_WithMQTT/fdrs_gateway_config.h:62: note: this is the location of the previous definition
   62 | #define TOPIC_COMMAND "proj/command"
```

This is my ino:
#include "fdrs_gateway_config.h"
#include <fdrs_gateway.h>

Stips5, 2024-06-12:
but when i set in my ino:
#include <fdrs_globals.h>
#include "fdrs_gateway_config.h"
#include <fdrs_gateway.h>

and in my fdrs_gateway_config

```

// MQTT Topics
#undef TOPIC_DATA
#define TOPIC_DATA "proj/data"

#undef TOPIC_STATUS
#define TOPIC_STATUS "proj/status"

#undef TOPIC_COMMAND
#define TOPIC_COMMAND "proj/command"
```

then it seems to work

aviateur17, 2024-06-13:
Created PR #216 for MQTT topics

```
// MQTT Topics
#define GLOBAL_TOPIC_DATA    "fdrs/data"
#define GLOBAL_TOPIC_STATUS  "fdrs/status"
#define GLOBAL_TOPIC_COMMAND "fdrs/command"
#define GLOBAL_TOPIC_DATA_BACKLOG "fdrs/databacklog"   // Used in filesystem module
```

timmbogner, 2024-06-17:
Hey @Stips5 ,
The changes are now live in the main branch. Thank you for making that suggestion! Please let us know if you face any more inconveniences like that, especially when expanding to multiple systems.

Stips5, 2024-06-20:
works, thanks