timmbogner/Farm-Data-Relay-System #233: unable to compile ...

urs8000, 2024-08-28:
in case of missing <sys/time.h>
why don't you use an universal time library for that functions?
e.g. Time with TimeLib.h ???

timmbogner, 2024-08-29:
Hi @urs8000,
There are issues with AVR boards not having the [sys/time.h](https://pubs.opengroup.org/onlinepubs/007908775/xsh/systime.h.html) file, but (to my knowledge) it should be present for SAMD and other boards/chips with time functionality. I am testing the 'LoRa_Sensor' sketch under the _Adafruit Feather M0 (SAMD21)_ definition and it compiles successfully. Let me know if your setup is different.

urs8000, 2024-09-01:
hello @timmbogner 
thank for your answer
I installed FDRS from scratch and could compile your unmodified example LoRa_Sensor without error (only some notes & warnings)
as soon as I activate DGB_Level 1 or 2 I get:
In file included from e:\_development\Arduino\libraries\Farm-Data-Relay-System\src/fdrs_node.h:40,
                 from E:\_development\Arduino\libraries\Farm-Data-Relay-System\examples\LoRa_Sensor\LoRa_Sensor.ino:10:
e:\_development\Arduino\libraries\Farm-Data-Relay-System\src/fdrs_time.h: In function 'void adjTimeforNetDelay(time_t)':
e:\_development\Arduino\libraries\Farm-Data-Relay-System\src/fdrs_time.h:391:45: error: call of overloaded 'String(time_t&)' is ambiguous
  391 |       DBG1("Time adj by " + String(newOffset) + " secs");

at the end I could compile HELTEC_LoRa_32(V2), Feather_M0(SAMD21)_LoRa and other ESP based controllers
but NOT ANY ARM based e.g. MoteinoMEGA(-USB)

to overcome this problems I've used a save library bundle dated of 26/02/2024 to compile onla the MoteinoMEGA-sensor

it would be great if you could EXCLUDE timeinformation from any sensor. that's a function one (best MQTT) of the gateways should fullfill.

closed 2025-02-10 by timmbogner