timmbogner/Farm-Data-Relay-System #243: Only to compile with Arduino IDE not with PlatformIo

hansratzinger, 2025-03-02:
I tried for two days to get the UART gateway example compiled with PlatformIo. As there is no library.json in the FDRS repo I uploaded the header files in the /lib/FDRS/src folder. But even with help of my coworker ;-) Copilot I was not able to compile it without errors. Although I dont like to work with the Arduino IDE I tried finaly to compile the code with it and … it worked without any error. 

platformio.ini

`[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = bblanchon/ArduinoJson@^7.3.1
`

compiler log

`Verbose mode can be enabled via -v, --verbose option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.10.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 7.3.1
|-- Farm-Data-Relay-System @ 2.0.1
|-- WiFi @ 2.0.0
Building in release mode
Compiling .pio\build\esp32dev\src\1_UART_Gateway.cpp.o
Compiling .pio\build\esp32dev\lib326\WiFi\WiFiMulti.cpp.o
Compiling .pio\build\esp32dev\lib326\WiFi\WiFiSTA.cpp.o
Compiling .pio\build\esp32dev\lib326\WiFi\WiFiScan.cpp.o
In file included from lib/Farm-Data-Relay-System/src/fdrs_gateway.h:80,
                 from src/1_UART_Gateway.cpp:10:
lib/Farm-Data-Relay-System/src/fdrs_gateway_serial.h:19:6: warning: #warning Defining RXD2 and TXD2 using MCU defaults. [-Wcpp]
     #warning Defining RXD2 and TXD2 using MCU defaults.
      ^~~~~~~
In file included from lib/Farm-Data-Relay-System/src/fdrs_gateway.h:80,
                 from src/1_UART_Gateway.cpp:10:
lib/Farm-Data-Relay-System/src/fdrs_gateway_serial.h: In function 'void getSerial()':
lib/Farm-Data-Relay-System/src/fdrs_gateway_serial.h:164:30: warning: 'bool ArduinoJson::V731PB22::JsonObject::containsKey(const TString&) const [with TString = char [5]; typename ArduinoJson::V731PB22::detail::enable_if<ArduinoJson::V731PB22::detail::IsString<TString>::value, int>::type <anonymous> = 0] is deprecated: use obj[key].is<T>() instead [-Wdeprecated-declarations]
     if(obj.containsKey("type")) { // DataReading
                              ^
In file included from .pio/libdeps/esp32dev/ArduinoJson/src/ArduinoJson.hpp:30,
                 from .pio/libdeps/esp32dev/ArduinoJson/src/ArduinoJson.h:9,  
                 from lib/Farm-Data-Relay-System/src/fdrs_gateway_serial.h:1, 
                 from lib/Farm-Data-Relay-System/src/fdrs_gateway.h:80,       
                 from src/1_UART_Gateway.cpp:10:
.pio/libdeps/esp32dev/ArduinoJson/src/ArduinoJson/Object/JsonObject.hpp:168:8: note: declared here
   bool containsKey(const TString& key) const {
        ^~~~~~~~~~~
In file included from lib/Farm-Data-Relay-System/src/fdrs_gateway.h:80,
                 from src/1_UART_Gateway.cpp:10:
lib/Farm-Data-Relay-System/src/fdrs_gateway_serial.h:178:34: warning: 'bool ArduinoJson::V731PB22::JsonObject::containsKey(const TString&) const [with TString = char [4]; typename ArduinoJson::V731PB22::detail::enable_if<ArduinoJson::V731PB22::detail::IsString<TString>::value, int>::type <anonymous> = 0]' is deprecated: use obj[key].is<T>() instead [-Wdeprecated-declarations]
     else if(obj.containsKey("cmd")) { // SystemPacket
                                  ^
In file included from .pio/libdeps/esp32dev/ArduinoJson/src/ArduinoJson.hpp:30,
                 from .pio/libdeps/esp32dev/ArduinoJson/src/ArduinoJson.h:9,  
                 from lib/Farm-Data-Relay-System/src/fdrs_gateway_serial.h:1, 
                 from lib/Farm-Data-Relay-System/src/fdrs_gateway.h:80,       
                 from src/1_UART_Gateway.cpp:10:
.pio/libdeps/esp32dev/ArduinoJson/src/ArduinoJson/Object/JsonObject.hpp:168:8: note: declared here
   bool containsKey(const TString& key) const {
        ^~~~~~~~~~~
In file included from lib/Farm-Data-Relay-System/src/fdrs_gateway.h:83,
                 from src/1_UART_Gateway.cpp:10:
lib/Farm-Data-Relay-System/src/fdrs_gateway_espnow.h: At global scope:        
lib/Farm-Data-Relay-System/src/fdrs_gateway_espnow.h:45:23: error: 'esp_now_recv_info' does not name a type; did you mean 'esp_now_peer_info'?
 void OnDataRecv(const esp_now_recv_info *pkt_info, const uint8_t *incomingData, int len)
                       ^~~~~~~~~~~~~~~~~
                       esp_now_peer_info
lib/Farm-Data-Relay-System/src/fdrs_gateway_espnow.h: In function 'void OnDataRecv(const int*, const uint8_t*, int)':
lib/Farm-Data-Relay-System/src/fdrs_gateway_espnow.h:47:29: error: request for member 'src_addr' in '* pkt_info', which is of non-class type 'const int'    
   memcpy(&incMAC, pkt_info->src_addr, sizeof(incMAC));
                             ^~~~~~~~
lib/Farm-Data-Relay-System/src/fdrs_gateway_espnow.h: In function 'void begin_espnow()':
lib/Farm-Data-Relay-System/src/fdrs_gateway_espnow.h:106:28: error: invalid conversion from 'void (*)(const int*, const uint8_t*, int)' {aka 'void (*)(const int*, const unsigned char*, int)'} to 'esp_now_recv_cb_t' {aka 'void (*)(const unsigned char*, const unsigned char*, int)'} [-fpermissive]
   esp_now_register_recv_cb(OnDataRecv);
                            ^~~~~~~~~~
In file included from lib/Farm-Data-Relay-System/src/fdrs_gateway_espnow.h:5, 
                 from lib/Farm-Data-Relay-System/src/fdrs_gateway.h:83,       
                 from src/1_UART_Gateway.cpp:10:
C:/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_wifi/include/esp_now.h:136:54: note:   initializing argument 1 of 'esp_err_t esp_now_register_recv_cb(esp_now_recv_cb_t)'
 esp_err_t esp_now_register_recv_cb(esp_now_recv_cb_t cb);
                                    ~~~~~~~~~~~~~~~~~~^~
In file included from lib/Farm-Data-Relay-System/src/fdrs_gateway.h:83,
                 from src/1_UART_Gateway.cpp:10:
lib/Farm-Data-Relay-System/src/fdrs_gateway_espnow.h: In function 'void add_espnow_peer()':
lib/Farm-Data-Relay-System/src/fdrs_gateway_espnow.h:202:63: warning: narrowing conversion of 'now' from 'time_t' {aka 'long int'} to 'uint32_t' {aka 'unsigned int'} inside { } [-Wnarrowing]
     SystemPacket sys_packet = { .cmd = cmd_time, .param = now };
                                                               ^
lib/Farm-Data-Relay-System/src/fdrs_gateway_espnow.h: In function 'esp_err_t sendTimeESPNow()':
lib/Farm-Data-Relay-System/src/fdrs_gateway_espnow.h:501:61: warning: narrowing conversion of 'now' from 'time_t' {aka 'long int'} to 'uint32_t' {aka 'unsigned int'} inside { } [-Wnarrowing]
   SystemPacket sys_packet = { .cmd = cmd_time, .param = now };
                                                             ^
lib/Farm-Data-Relay-System/src/fdrs_gateway_espnow.h: In function 'esp_err_t sendTimeESPNow(uint8_t*)':
lib/Farm-Data-Relay-System/src/fdrs_gateway_espnow.h:526:61: warning: narrowing conversion of 'now' from 'time_t' {aka 'long int'} to 'uint32_t' {aka 'unsigned int'} inside { } [-Wnarrowing]
   SystemPacket sys_packet = { .cmd = cmd_time, .param = now };
                                                             ^
*** [.pio\build\esp32dev\src\1_UART_Gateway.cpp.o] Error 1
======================== [FAILED] Took 11.94 seconds ========================

`

hansratzinger, 2025-03-03:
[PlatformIo Community Platform](https://community.platformio.org/t/farm-data-relay-system-not-able-to-compile-in-platformio-but-in-arduino-ide/46395?u=gruenthal99)

After many hours and help from the community (see above) I found out, that inspite of editing the .cpp files (former .ino) and putting #include <Arduino.h> on top, it was important to edit also every header file in the library in the same way. This was the only reason and am very happy beeing able to include the FDRS library in my PlatformIo based work.

closed 2025-03-03 by hansratzinger

aviateur17, 2025-03-03:
I hope that is not true.  I use PlatformIO as well and modifying every file does not seem like the way to go.  Looks like it's possibly some change in the ArduinoJson library.   I haven't updated and compiled lately so I will check this out but it may not be for a few days.

timmbogner, 2025-03-23:
@aviateur17 I hadn't seen your reply until now, but I've been curious about this. 
Is there something going on here that we could fix for PIO compatibility? I've been away from electronics mostly, but had to use Arduino for an unrelated project recently... It has inspired me to look into PIO next time I'm getting into things.

aviateur17, 2025-03-23:
I haven't had a chance to look at this.  Hopefully tonight but can't guarantee.  Will report back.

aviateur17, 2025-03-28:
@hansratzinger , successful VSCode/PIO compile with steps below and platformio.ini file below.  Next will try with your updated AduinoJson version.

1. Copy files from repo src folder into PIO project folder, src subdirectory
2. Copy example files into PIO project folder, src subdirectory
3. Create platformio.ini file in PIO project root folder (contents listed below)
4. rename .ino file to main.cpp
5. in main.cpp add #include <Arduino.h> to the top of the file.
6. Open PlatformIO terminal
7. 'pio run' (without quotes)

```
[env:esp32_7488f0]
platform = ${common.platform}
platform_packages = ${common.platform_packages}
board = wemos_d1_mini32
framework = arduino
monitor_speed = 115200
lib_ldf_mode = ${common.lib_ldf_mode}
build_flags = 
	${common.build_flags}
	-DHOST7488F0
board_build.partitions = min_spiffs.csv
lib_deps = ${common.lib_deps}
board_build.f_cpu = ${common.board_build.f_cpu}
board_build.f_flash = ${common.board_build.f_flash}

[common]
platform = https://github.com/platformio/platform-espressif32.git
platform_packages=
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.1
  framework-arduinoespressif32-libs @ https://github.com/espressif/arduino-esp32/releases/download/3.0.1/esp32-arduino-libs-3.0.1.zip
lib_ldf_mode = deep
build_flags = 
	-DCORE_DEBUG_LEVEL=5
lib_deps = 
	knolleary/PubSubClient@^2.8
	jgromes/RadioLib@^6.3.0
	bblanchon/ArduinoJson@^7.0.3
	makuna/RTC @ ^2.3.7
	adafruit/Adafruit NeoPixel @ ^1.12.0
board_build.f_cpu = 240000000L
board_build.f_flash = 80000000L
board_build.partitions = min_spiffs.csv

[platformio]
description = Arduino FW ESP32 FDRS Test
default_envs =  esp32_7488f0
```

```
pio run
Processing esp32_7488f0 (platform: https://github.com/platformio/platform-espressif32.git; board: wemos_d1_mini32; framework: arduino)
---------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/wemos_d1_mini32.html
PLATFORM: Espressif 32 (6.10.0+sha.33c5d50) > WEMOS D1 MINI ESP32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.0.1+sha.cc50d90 
 - framework-arduinoespressif32-libs @ 5.1.0+sha.442a798083 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep, Compatibility ~ soft
Found 43 compatible libraries
Scanning dependencies...
Dependency Graph
|-- PubSubClient @ 2.8.0
|-- RadioLib @ 6.6.0
|-- ArduinoJson @ 7.3.1
|-- RTC @ 2.5.0
|-- Adafruit NeoPixel @ 1.12.5
|-- Wire @ 2.0.0
|-- WiFi @ 2.0.0
|-- ArduinoOTA @ 2.0.0
|-- Ethernet @ 2.0.0
Building in release mode
Compiling .pio/build/esp32_7488f0/src/ESP8266_and_ESP32_OLED_driver_for_SSD1306_displays/src/OLEDDisplay.cpp.o
Compiling .pio/build/esp32_7488f0/src/ESP8266_and_ESP32_OLED_driver_for_SSD1306_displays/src/OLEDDisplayUi.cpp.o
Compiling .pio/build/esp32_7488f0/src/main.cpp.o
Building .pio/build/esp32_7488f0/bootloader.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Generating partitions .pio/build/esp32_7488f0/partitions.bin

..... compile continues.....

Compiling .pio/build/esp32_7488f0/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/esp32_7488f0/libFrameworkArduino.a
Indexing .pio/build/esp32_7488f0/libFrameworkArduino.a
Linking .pio/build/esp32_7488f0/firmware.elf
Retrieving maximum program size .pio/build/esp32_7488f0/firmware.elf
Checking size .pio/build/esp32_7488f0/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  13.5% (used 44296 bytes from 327680 bytes)
Flash: [=====     ]  48.1% (used 945285 bytes from 1966080 bytes)
Building .pio/build/esp32_7488f0/firmware.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 27 ELF sections
Successfully created esp32 image.
========================= [SUCCESS] Took 77.54 seconds =========================
```

aviateur17, 2025-03-28:
@hansratzinger, commented out my platform and platform_packages platformio.ini lines and just used espressif32 and compile failed just as yours did.  Espressif must have changed their API for esp_now in their new packages.  We'll need to correct for that.    For now, until that is corrected, use the platform and platform_packages lines in my sample platformio.ini file in the previous post.  Sorry for the inconvenience.

hansratzinger, 2025-03-29:
@aviateur17, thx so much to make it clear!