timmbogner/Farm-Data-Relay-System #246: Add COUNTER data type

PerRieland, 2025-03-27:
Please add some counters to the data type.
I'm about to count Bluetooth/WiFi devices nearby nodes/controllers and missing counters for that. 
Maybe it could be COUNTER_T, COUNTER2_T

hansratzinger, 2025-03-29:
@PerRieland You can modify the data types to your needs. I figured out that the data types have no affect to the function of the code. All of them are of the type float and you can redefine the data types as you want. I did it in my project like this:

// FDRS Data Types
#define STATUS_T                     0  // Status 0 = OK, 1 = Error
#define DATE                           01
#define UTC                             02
#define BOARDTIME_MILLSEC 07 // millis() from start of the board
#define BOARDTIME_SEC        16  // secounds,millsec
#define DIRECTION_LAT         18 // direction lat 1/N 0/S 
#define DIRECTION_LON        19 // direction lat 1/N 0/S 
#define RPM                            20 // RPM
#define LATITUDE                    21 // GPS Latitude
#define LONGITUDE                22 // GPS Longitude
#define ALTITUDE_T                 23 // GPS Altitude
#define HDOP_T                      24 // GPS HDOP
#define SATELLITES                  25 // satellites
#define SPEED_KMH                26 // SPEED KMH
#define HEADING                    27 // heading
#define POSITION_DISTANCE  28 // positionDifference

PerRieland, 2025-03-29:
Yes @hansratzinger you are right. But I didn't want to use some (random) ID, that later on will be used in FDRS. 
For the time, I'm using my own (random9 ID, and it works just perfect. ;)

timmbogner, 2025-04-01:
Hi Per,
The best way to get a new type added is generally to just create a pull request with the desired change. The man with the Swiss accent has a [great tutorial](https://www.youtube.com/watch?v=LtA7GQYQzjI) on it if that's unfamiliar territory. IT_T was intended as the counting type, but I think COUNTER_T (and 2) would be different enough to warrant an addition.
Thanks! 

I'm going to move this to "discussions".